knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #157: First layer too high

kahoch, 2021-06-06:
[image]
This is my home position, the nozzle is a little bit higher than the plate(Why, because I don't want the nozzle press the plate in every homing cycle). 
My clone BLtouch sensor and auto leveling works fine, and I had added M420 S1 in the start code, but first layer always too high. 
It seems the machine cannot go to a negative Z height during printing. 
How can I fix this?
Thank you!

Jack77777777, 2021-06-06:
If you manual level your bed with a gap like this to the nozzle and not like described in the wiki you have to disable software endstops M211 S0. Otherwise the nozzle will never lower more then to the endstop of the sensors.

Jack77777777, 2021-06-06:
Don't forget to enable it afterwards M211 S1. Printer reboot will enable it too.

kahoch, 2021-06-07:
> 
> 
> If you manual level your bed with a gap like this to the nozzle and not like described in the wiki you have to disable software endstops M211 S0. Otherwise the nozzle will never lower more then to the endstop of the sensors.

Thank you Jack! I didn't see any description in wiki about this. :dizzy_face:
I'll try to add M211 S0 to the start code and S1 to the end code

Jack77777777, 2021-06-07:
Don't add it to the start or end gcodes. You only have to use it to determine your offset and then add it with M851 in terminal and save with M500. You can check the value with M503

kahoch, 2021-06-08:
> 
> 
> Don't add it to the start or end gcodes. You only have to use it to determine your offset and then add it with M851 in terminal and save with M500. You can check the value with M503

OK， I decide to manual level again. Make the nozzle touch the plate. Maybe I'll cut a notch on the plate later.:stuck_out_tongue_closed_eyes:
Thank you again!

Jack77777777, 2021-06-08:
Use a bill (Kassenzettel 0.05mm) between nozzle and bed when leveling