knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #158: [BUG] BLtouch Mega x not working with beta 4

MaGe-1, 2021-06-06:
When starting Probing via the Special Menu with the mega X and Beta 4, the pins moves in and out 4 times. After that, the printer will move to the error state. Message "Bltouch error"

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2021-06-07:
Is this your code?

`#if ENABLED(KNUTWURST_BLTOUCH, KNUTWURST_CHIRON)
#define Z_MIN_PROBE_PIN 21 // Pin 32 is the RAMPS default
#endif`

knutwurst, 2021-06-07:
Did you update from a release version (e.g. 1.1.9) or is the 1.2.0 Beta 4 the only one you tried?

MaGe-1, 2021-06-07:
> Did you update from a release version (e.g. 1.1.9) or is the 1.2.0 Beta 4 the only one you tried?

I Updated from beta 3, where this functionality worked.
I changed the Signal Pin from pin 2 to pin 21. This is the pin that you used in one of your Youtube Videos. 

The Rest of the Code is unchanged from the downloade zip file

MaGe-1, 2021-06-07:
My guess is, that there might be an issue with the New line "KNUTWURST_CHIRON". Is it possible, that my changed pin is ignored in the build process? 

Thanks for your great work btw

MaGe-1, 2021-06-13:
With beta 2 ABL is working

JernejL, 2021-06-14:
I have a mega x and got bltouch working properly on first try in beta 4. 
I used MEGA_X_DGUS_BLT_10_v1.2.0-beta.4 and bltouch is a clone. 
 
See https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/149 there are also pictures how i wired it.

MaGe-1, 2021-06-14:
So you did Not build the Firmware yourself?

JernejL, 2021-06-14:
> So you did Not build the Firmware yourself?

Sorry, you are correct - i didn't notice that you have changed the pin assignments and rebuilt it.

knutwurst, 2021-06-14:
@MaGe-1 , can you tell me the exact line where you changed the pin assignment? It should be fine. The Chiron uses the same pin "2", and if you change it to 21, it should work out of the box.

Can you show some pictures of your wiring?

MaGe-1, 2021-06-14:
Code I used in beta 2:
#if ENABLED(KNUTWURST_BLTOUCH)
    #define Z_MIN_PROBE_PIN 21 // Pin 32 is the RAMPS default
#endif

In beta 4 i did the same, But IT did Not work. Maybe the pin change is ignored in the build process?

I connected the bltouch like in your Youtube Video from the mega x and connected the bltouch direct to the motherboard. 
[image]

MaGe-1, 2021-07-14:
Hi, I still have the Problem with the newer version.

MaGe-1, 2021-07-14:
i can confirm that the following is fixing the issue:

#if ENABLED(KNUTWURST_BLTOUCH, KNUTWURST_CHIRON) <-- delete this line
#define Z_MIN_PROBE_PIN 21 // Pin 32 is the RAMPS default
#endif <-- delete this line

its somehow not compiled into the fw of mega x....

knutwurst, 2021-07-30:
The pin assignment moved the the pins_RAMPS.h, so there is no need to add something in the configuration.h.

closed 2021-07-30 by knutwurst