knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #159: [BUG] BLTouch Clone (3DTouch) not triggering - Trigorilla v0.0.3

ksmarty, 2021-06-07:
### Bug Description

I am unable to get the 3DTouch sensor to trigger as per step 9 in the BLTouch installation manual.

### Steps to Reproduce

I have an Anycubic Mega S with a Trigorilla v0.0.3 board. I flashed the MEGA_S_DGUS_BLT_10_v1.1.9 firmware. Both the blue and red lights are on and it clicks up and down on startup. I followed the steps in the BLTouch installation manual. Once the pin touched the bed, nothing happened. I didn't get a notification in the terminal.

I found a similar issue but the user didn't post their solution or a link to the "italian sites": #70

The main issue I believe is that there is no S0, only S1 on this board, but I'm not sure why this isn't reported a touch.

knutwurst, 2021-06-07:
S0 or S1 is only the Servo-Port. If the probe extends when starting (G29 or via the Special Menu), the sensor connection is correct! The trigger is the two (black and white) cables that you solder to the hotend board.

Pay close attention to the wiring. The two must not be swapped either.

However, I have already seen that there are other hotend boards on which SMD capacitors are soldered. These are useless and must be unsoldered. Otherwise the BLT will not work.

ksmarty, 2021-06-07:
I just removed the hotend board and checked my connections. Both B5 and B6 are soldered correctly and have continuity to the corresponding pins on the other side. I made sure to plug the cable in correctly, but still no message in the terminal. I also checked the voltage on the black and white wires when the sensor is activated and it does report a voltage. 

One thing I did notice in testing was that B6 and A6 have continuity. Is this supposed to be the case?

I'm not really sure what else I can check at this point.

knutwurst, 2021-06-07:
Can you post a picture of your hotend pcb?

ksmarty, 2021-06-07:
I know my soldering job is not great (I'm buying a proper kit today), but all of my connections seem to be solid.

[image]
[image]

closed 2021-07-14 by github-actions[bot]