knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #160: [BUG] Mega x beta 4 - minor LCD glitches on blue/yellow

JernejL, 2021-06-14:
### Bug Description

I have a Mega X with blue-yellow lcd, middle picture here: 
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Types-of-Anycubic-Touchscreens 
 
Knutwurst fw is: MEGA_X_DGUS_BLT_10_v1.2.0-beta.4 
 
Special menu works, but sometimes ( seems like most of time) will not activate features until i click on item, then click print nothing happens, but when i afterwards click refresh the feature is triggered normally. 
 
I'm not sure, if this is the problem that the mentioned LCD firmware update fixes? 
 
I can make a video of this, let me know if you need anything tested or if the firmware update is what will solve this. 
 
I intend to upgrade printer to octoprint in future, can the printer run without the LCD and i just use USB interface? 
 
### Steps to Reproduce

1. Enter special menu
2. Select item
3. Nothing happens

**Expected behavior:** Feature is triggered

**Actual behavior:** Feature is not triggered, but is after refresh is pressed.

JernejL, 2021-07-04:
Adding a video of this: 
 
https://www.youtube.com/watch?v=JMXyaWHZYEM

knutwurst, 2021-07-04:
It's not a bug. To trigger a feature you have to click on "reload" since the "print" button is tied to the "print the selected file" behaviour. 

I'm sorry I didn't write this in the instructions. I wanted to write a beginner's guide anyway.

Also, there is the 1.2.0 Final out now: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases

closed 2021-07-04 by knutwurst

JernejL, 2021-07-04:
AAAH so basically.. reload replaces the print / run button.. i got it. 
 
Thanks, and yeah might make sense to update instructions. 
 
Btw, i posted some good pictures for bltouch wiring on mega x here: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/149 - it could be of use to you in the wiki for guides, feel free to use those pictures. 
 
I will give 1.2.0 a try soon after i finish a few prints.

knutwurst, 2021-07-04:
> AAAH so basically.. reload replaces the print / run button.. i got it.
Not exactely ... only in the special menu, reload is the select button, but when you select a file from sd card, you need to press "print" like in the stick firmware. 

This had to be a woraround, since the display itself has some internal logic which can not be altered.

JernejL, 2021-07-04:
I guess octoprint is what i should go for in the future, and nuke the original display.