knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #162: [FR] Version for Mega S with BMG extruder

condor125, 2021-06-17:
### Description

I have the Mega S with BMG extruder, could you please make the firmware available for this combo? Thanks!

knutwurst, 2021-06-17:
The Feature is already in there. You only need to compile it by yourself with PlatformIO and set the environment variable **-DKNUTWURST_BMG**

Jack77777777, 2021-06-18:
I have a BMG Clone from Trianglelab and just changed the E-Steps through terminal. Should i change anything in the firmware and compile ?

knutwurst, 2021-06-18:
@Jack77777777 if your Extruder runs in the correct direction, then no... you don't need to. But with the Flag, everything is done for you... correct direction for your printer, speed, gear reduction etc.

Just add it to your printer config and compile.
[image]

Jack77777777, 2021-06-18:
I swapped the pins to correct the direction. Would you recommend to swap back and let the firmware do it ?

knutwurst, 2021-06-21:
@Jack77777777 you can do whatever you like :) I doesn't make any difference at all. :)

closed 2021-06-21 by knutwurst