knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #163: [BUG] Filament Feed Reversed

hubortje, 2021-06-18:
### Bug Description

After flashing MEGA_P_TMC_v1.1.9.hex, the filament feeder is reversed. While printing filament is ejected, and when using the "remove filament" option in the menu, filament is fed into the nozzle.

### Steps to Reproduce

1. Flash MEGA_P_TMC_v1.1.9.hex with CURA via USB.
2. Start printing a .gcode-file via Octoprint

**Expected behavior:** Printing ensues like usually.

**Actual behavior:** While X, Y, Z, work as expected, the filament motor seems to be reversed.

knutwurst, 2021-06-21:
Hi, did you change ALL of the stepper drivers to TMC2208? Or are you just using the stock Mega Pro?

hubortje, 2021-06-23:
I'm on it. Bought it used from a friend, asked him and am waiting for his answer.

knutwurst, 2021-06-29:
You could just open the printer and take a photo of your motherboard :)

hubortje, 2021-06-29:
Thank you for the tip, but I've got an answer already: As expected, the extruder is an A4988, not a TMC in contrary to the rest of the drivers.

hubortje, 2021-07-05:
Hey Knutwurst,

ich switch mal auf Deutsch, wenn es genehm ist.
Ich hab mir jetzt Visual Studio Code eingerichtet. Muss ich im Configuration-Header lediglich in Zeile 804 aus
`#define E0_DRIVER_TYPE TMC2208_STANDALONE` TMC2208_STANDALONE durch A4988 ersetzen?
Oder muss ich noch an einem Punkt schauen, dass ich das richtig ergänze?

knutwurst, 2021-07-05:
Hallo :) 
Ja, ist ok - auch wenn Github eher eine komplett englische Sache ist und ich hier gern auf englisch kommuniziere, weil es eben auch dann international weiterhilft.

Auf welchem Stand des Sourcecodes arbeitest du denn? In 1.1.9 hatte ich die Timings noch selbst gebaut und als Treiber immer die A4988 definiert (weil's egal war). Der Sourcecode der jeweilien Version ist als Download bei den Assets zu finden. Wenn du den Master gecloned hast, bist du auf einer Dev-Version 1.2.1, in der immer wieder was verändert wird. Also verlass dich bitte nicht drauf, dass die auch läuft.

Ab der 1.2.0 bin ich wegen Beschwerden beim originalen Mega Pro zurück zur Unterscheidung zwischen TMC22xx/A4988 gegangen. 

> Muss ich im Configuration-Header lediglich in Zeile 804 aus
#define E0_DRIVER_TYPE TMC2208_STANDALONE TMC2208_STANDALONE durch A4988 ersetzen?

Also im Master ist das diese Zeile. Und ja, das ist korrekt :)
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/b740c9f747de924013d7f280257080f0b63a4a2c/Marlin/Configuration.h#L804

Den musst du auf jeden Fall ändern, so dass die Timings stimmen. Allerdings musst du jetzt noch die Logik drehen. Das geht am einfachsten, wenn du das KNUTWURST_BMG Flag nutzt. Das setzt die richtigen Beschleunigungswerte und invertiert die Drehrichtung. Das kannst du für jede Konfiguration individuell in der i3_mega_envs.ini setzen und dir so dein Profil zusammenbauen.

Es reicht, wenn du das einfach aus der Zeile rauslöschst. Mehr musst du nicht machen. Einfach mit dem MEGA_P_TMC Profil bauen und fertig.
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/b740c9f747de924013d7f280257080f0b63a4a2c/i3_mega_envs.ini#L412

Ich hoffe ich konnte dir helfen ;)

hubortje, 2021-07-26:
Fair enough, then let us switch back to english (finest denglish thread).

So, I changed line 804 in the configuration.h accordingly, and built the .hex-file. I flashed it via cura. The printer did not restart, but instead of restarting, the printer just beeped. A test via Filament In resulted in a removal of the filament.
I restarted the printer manually and tried again, [same result].

I'm not sure if I correctly understood what you mean with the KNUTWURST_BMG flag, as it is already set in the profile you mentioned? I will check the settings again. As platformio has problems running on my laptop and my computer, I have to do it via my girlfriends laptop, kinda tedious. :-)

Thank you for your help so far. Did not have a lot of time on hand, but now I've got a few months to dig in! :-)

knutwurst, 2021-07-27:
When the printer makes the start-up sound, it will restart. The display never goes out because it is still supplied with power and does not have its own reset.

It is so:
5 drivers are built into the normal Mega Pro.
3x Polulu A4988 for Z0, Z1 and E0
2x Trinamic TMC2208 for X and Y

The profile "MEGA_P" activates exactly this configuration.

If you now also exchange the last 3 drivers for TMC2208 you have 5x TMC2208. So for X, Y, Z0, Z1 and E0.

This is mapped in the MEGA_P_TMC profile.

In your case, as I understand it, it looks like this:
4x Trinamic TMC2208 on X, Y, Z0, Z1
1x Polulu A4988 on E0 (for whatever reason. Maybe he didn't have another driver anymore)

So you have two options:

1. You look for the MEGA_P entry in the source code and also change the drivers on Z0 and Z1 ... or ... for the MEGA_P profile.

2. You look for the MEGA_P_TMC entry in the source code and just change the E0 driver back to A4988.

> I'm not sure if I correctly understood what you mean with the KNUTWURST_BMG flag, as it is already set in the profile you mentioned?

The BMG extruder, which was built into the Mega Pro at the factory, has the property of rotating exactly the other way around than the MK8 or titanium extruder. That is why there is the BMG flag, which is set on the MEGA_P and MEGA_P_TMC - precisely because a BMG is installed.

With other printers, the flag also changes acceleration values and steps, but with the Mega P the flag only changes the direction of rotation of E0. If your extruder is now running the wrong way round, you can simply delete the flag from the ini file.

How you solve it is up to you.
The second solution seemed easier to me.

hubortje, 2021-07-27:
Thank you again for your detailed explanation, helps a lot to understand!
Now I understood, what you meant with the BMG_Flag, as I did not grasp to **delete** the flag out of the line.

Though I realized a grave rookie mistake from my side. I'm not running on a Mega P, but just an Anycubic i3 Mega.
Does the deletion of the flag still applies if I build the MEGA_TMC profile?

knutwurst, 2021-07-27:
Now I'm confused.
What kind of printer do you have now?
A Mega? Mega S? Mega P?
Which extruder is installed? Which drivers?

hubortje, 2021-07-27:
Story of my life.

I have an Anycubic i3 Mega (confirmed via the About menu)
I have the standard extruder (A4988), otherwise TMC drivers built in.

knutwurst, 2021-07-27:
The standard Extruder (MK8 Clone)? ... A4988 ist the Stepper Motor Driver.
Can you post a photo?

hubortje, 2021-07-27:
Sure thing:
[image]
[image]

knutwurst, 2021-07-27:
Okay.. That's the i3 Mega, but with Mega S glass bed.

Do you have one or two Z-endstop switches?

If you have two endstops, you need to do the following:
Edit this to "A4988":
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/b3f4dcb2d1bcf2f6f3148945852d49edc2d35106/Marlin/Configuration.h#L804

And here set the entry to "false":
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/b3f4dcb2d1bcf2f6f3148945852d49edc2d35106/Marlin/Configuration.h#L1445

Then you should build the Environment **MEGA_TMC** without any issues. Please keep in mind that you need to load the FW defaults after flashing.

Please also forget about the KNUTWURST_BMG flag, since you don't have a BMG (bondtech mini geared) extruder. ;)

hubortje, 2021-07-27:
Thank you very much for the info and explanation!

Where can I check if I've got two Z-endstop switches?

Regarding the glass bed, does this change the recommended sizes?

knutwurst, 2021-07-27:
No problem :) 

Uh... just look for micro-switches on the left and right side of both Z motors near the glass bed. They are always paired with a long screw which presses against the lever.

> Regarding the glass bed, does this change the recommended sizes?
Yes and no. On the normal i3 Mega, the Bed is a bit smaller (210x210mm). If you flash the MEGA firmware, your print volume stays the same and you don't need to set up anything. You just can't use the full bed as printable area. On the Mega S and P I increased the build volume to make use of the bigger bed - but the Mega S firmware also changes the acceleration settings and the extruder values.... so you should be good to go with the i3 Mega standard firmware.

hubortje, 2021-07-27:
I've got two! :-)

Built and flashed without problems, Printer prints happily away.
Thank you very much, Knutwurst!

closed 2021-07-27 by hubortje