knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #164: I3 Mega mit BL Touch hängt sich auf

Dogfroster, 2021-06-20:
Wie oben geschrieben hängt sich mein I3 Mega mit BLTouch beim Druckstart auf.

Ablauf: 
Druck wird gestartet, Bett heizt auf, BlTouch fängt an zu messen (16 Punkte) an der letzten Position fährt er noch 10 mm hoch.
Dann sollte er dei Nozzle aufheizen aber ab da treten ab und zu probleme in der art auf das der Drucker sich aufhängt.
1 bleibt nach dem Leveln hängen bevor die Nozzle aufgeheizt hat,
2 fängt an die Purgeline zu drucken und bleibt mitten drin stehen
3 mitten im druck bleibt er stehen.

Das Problem tritt auf egal ob ich mit Octoprint oder direkt von der Karte drucke.
Tritt mit der fertigen Version auf aber auch bei der selbstkompilierten (Dort eigentlich nur G26 deaktiviert, und die E Steps angepasst wegen Titan Extruder)

Board ist 1.0 Version und die Treiber die Original A4988.
Hab im Moment wieder die alte von D Ramiro neukompeliert und drauf gemacht weil der Drucker damit läuft er.

knutwurst, 2021-06-21:
Tja, so ohne weitere Informationen kann ich dazu auch nicht viel sagen, sorry. Hast du mal die FAQ gelesen? Du bist ja nicht der Erste mit dem Problem :)

Dogfroster, 2021-06-28:
Fehler gefunden war eine lose Schraube am Mosfet jetzt gehts.

closed 2021-06-28 by Dogfroster

knutwurst, 2021-06-28:
Dazu hatte ich auch mal was geschrieben ;) 
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Soll-ich-externe-MOSFETs-einbauen%3F