knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #165: Tmc2225, default 32 subdivision?

WenKwen, 2021-06-21:
Tmc2225, default 32 subdivision?
Due to the customized screen of anycubic-i3-mega-s

It is difficult to edit Marlin 2.0. X

Now the default 32 subdivision of tmc2225 causes that your firmware cannot be used

Can you give me a tutorial or 32 subdivision of mage-s TMC dgus firmware

The problem is: the sprinkler stopped before returning to the limit switch, and failed to hit the limit switch

WenKwen, 2021-06-21:
The problem is: the nozzle stopped before returning to the limit switch and failed to hit the limit switch

WenKwen, 2021-06-21:
Use 4988 normal use

Replace tmc2225 and confirm that the line sequence is correct and cannot return to the limit switch

knutwurst, 2021-06-21:
There is only one Atmega2560 with 16Mhz on the Trigorilla mainboard. This would not create a 1/32 microstepping. In my opinion, you only have the option of either exchanging the drivers for TMC2208 or 2209, or disconnecting the MS1 pin (see: https://wiki.fysetc.com/TMC2225/) and connecting it to ground.
[image]

WenKwen, 2021-06-21:
[image]

WenKwen, 2021-06-21:
thank you,Thank you for your answer

closed 2021-06-21 by knutwurst