knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #166: Kurz vor dem Druck spiegelverkehrtes Homing

lisi2001, 2021-06-21:
Hi,

ich habe seit kurzem die Version 1.2.0 beta 4 laufen, nachdem sich die letze stable version irgendwie verabschiedet hat. 
Fakt ist, die 1.2.0 b4 läuft perfekt mit dem BLTouch und dem Trigorilla 1.0 einschließlich den TMC. Leider kommt es bei meinen letzten beiden Ausdrucken (unterschiedliche Objekte) gesliced mit Cura 4.9.0 zu merkwürdigen Fehlern.

Der Drucker (Anycubic I3 Mega S) macht ein Homing - beginnt mit dem Autoleveln und beendet dieses 1A (von links vorne nach rechts hinten). Laut Konsole speichert er die MeasurePoints. 
Dann fährt er Z hoch, bewegt sich zu X, Y auf 0 und dann beginnt Y gefühlt im Schneckentempo sich auf 195 zu bewegen. Anschließend bewegt sich X im Schneckentempo auf 210. 

An diesem Punkt fährt er Z gegen das Bett und beginnt mit dem Extruden.

Die Konsole gibt den Normalen Output, also Homen und dann fährt er eigentlich den 1 Punkt an. Wie gesagt, den erreicht er nicht, da er irgendwie die gegenüberliegende Seite des Nullpunkts anfährt und dort beginnen möchte. 

Hat jemand solch einen ähnlichen bemerkt, bzw. gibt es hierzu einen Workaround zum solven?

Hilfe wäre nett bzw. vielleicht hab ich einen Bug gefunden.

Danke und liebe Grüße 
Lisbeth

knutwurst, 2021-06-29:
Hallo Lisbeth,
was bedeutet denn "nachdem sich die letze stable version irgendwie verabschiedet hat"? Eine Firmware kann sich eigentlich nicht 'verabschieden'. Das ist fester Hardware-Programmcode. Der ist unveränderlich.

Dein Fehler ist mir unbekannt und kann durch verschiedene Faktoren beeinflusst werden:
- Der Motorstrom ist zu niedrig bei verbauten TMC2208 Treibern -> Vref erhöhen (mind. 1V) oder verringern (falls über 1,4V)
- Die Versorgungsspannung ist allgemein nicht stabil -> Netzteil mit Oszilloskop prüfen oder austauschen
- Das EEPROM wurde nicht initialisiert und es werden völlig falsche Zufallswerte geladen -> "Load FW Defaults" im Special Menu

Falls du keine TMC Treiber verbaut hast, kann es sein, dass nicht näher spezifizierte A4988 Treiber mit meinen Timings nicht klarkommen und dann falschherum drehen.

Bitte installiere mal die neueste Final-Version 1.2.0. Hier wurde einiges verbessert und behoben.

lg
Olli

knutwurst, 2021-07-30:
Besteht das Problem noch?

closed 2021-08-05 by github-actions[bot]