knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #168: [BUG] (Thermistor temperature error)

DaniLeyend, 2021-06-28:
### Bug Description

The Hotend Temperature is displayed at 300º constantly.

### Steps to Reproduce

1. Download the source code. (Leatest version)
2. Change the Configuration.h file in line 488 #define TEMP_SENSOR_0 **1047** according to my thermistor
https://es.aliexpress.com/item/32859917151.html (it is the correct value)
4. Compile the firmware with Platformio in Microsoft Visual Studio
5. Flash the firmware (Chiron .hex) with XLoader to my 3D Printer Anycubic Chiron

**Expected behavior:** 

The thermistor marks the correct temperature Approx 20º same as the bed

**Actual behavior:** 

The Hotend thermistor temperature is displayed at 300º constantly.

#### Additional Information
I have tried other firmware and the thermistor temperature was working fine with this setting. (By the way, this is by far the best firmware and the most updated I have seen, I do not know how I have been able to live without it so far, but need to set up correctly that seeting).
Help please, I am not sure if i am messing up with something

[image]
[image]
[image]
[image]
The last 2 files are my Configuration.h and Configuration_adv_h renamed to .png if necessary

DaniLeyend, 2021-06-28:
I just changed the thermistor for a different one, I have recompiled the firmware and apparently everithing works fine

closed 2021-06-28 by DaniLeyend

knutwurst, 2021-06-29:
I'm glad that everything works fine ;)