knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #169: [BUG] (short description)

Mrinmoyst1, 2021-07-03:
### Pause and Resume printing isn't working at all....

I paused my print after the filament ran out and then fed the new one. After that when I pressed continue, only the monitor was saying that the printer is running but the nozzle didn't move at all. No progress bar was seen. It disappeared totally. I think there are some bugs. Please solve the issue. Also try to introduce resume printing after power loss. I had this feature in the stock firmware and it worked like a charm. Your firmware is really great and better than any others currently available. Please introduce this feature and solve the issue as soon as you can.

JernejL, 2021-07-24:
pausing and resuming is broken completely, my mega x will never offer to resume after pausing, it shows the "command sent" message forever.

knutwurst, 2021-07-30:
This is a duplication. Please see #34 .

closed 2021-07-30 by knutwurst