knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #171: [FR] Live Z offset adjustment

mrr3n3w, 2021-07-06:
### Description

Using the special menu the ability to live adjust z offset similar to that on newer creality firmware/ new anycubic firmware on the vyper.

### Feature Workflow

1. Open Special Menu
2. Z offset adjust
3. + / - Buttons
4. save to firmware button
5. During print allow for z offset adjust

#### Additional Information
as shown in this article under the screen section.
https://3dprintbeginner.com/anycubic-vyper-review/

MaGe-1, 2021-07-14:
i would love this too.
Would be nice for leveling.

knutwurst, 2021-07-24:
"Live" Z-Offset isn't possible, because the special Menu is not accessible while printing but normal Z-Offset setting would be possible. If this helps, I'll add it to the next release :)

mrr3n3w, 2021-07-24:
The special menu is available while printing using octoprint or via pc though... Just a thought

knutwurst, 2021-07-24:
Yes, ok, good point. But to me honest - I don't know if it works correctly *while* printing. At the moment it's still difficult to add this feature to the Chiron implementation, which has a menu entry in the display.

mrr3n3w, 2021-07-24:
Totally fair

I super appreciate everything you guys are doing. Keep up the good work.

closed 2021-08-29 by github-actions[bot]