knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #174: [BUG] (short description)

v-a-d-e-r, 2021-07-11:
BLTouch geht immer noch nicht. Nach dem Einschalten initialisiert er (wenn ich Glück habe...) 1 von 10 mal, ansonsten blinkt er! Wenn er das Mesh dann mal anfängt zu scannen, hört er nach 5 Abtastungen in der rechten unteren Ecke auf und das war es dann. Also alles eine einzige Katastrophe.

Die Daten:
- BLTouch orig. v3.1
- FW 1.2.0  DGUS+TMC+BLT+1_0
- Ja, die Hardware ist richtig angeschlossen (ist ja mein Job)
- Anycubic Mega S

MaGe-1, 2021-07-14:
Die Daten:

    BLTouch China. v3.1
    FW 1.2.0 DGUS+TMC+BLT+1_0
    Ja, die Hardware ist richtig angeschlossen (ist ironischerweise auch mein Job ;-P Aber Wichtiger: mit Beta 2 geht es)
    Ja, Reset habe ich gemacht
    Anycubic Mega X
    
    Verhalten: Bei Auslösen der Kalibrierung geht der BLTOUCH 3x raus und wieder rein -> Danach folgt "Error wegen BLTOUCH, Restart with M..."

sknierimm, 2021-07-23:
Ich hänge mich da mal ran:
Mega X - vom Setup her exakt genau so, wie im Wiki beschrieben. Auch bei mir läuft die Prozedur mit dem Offset ermitteln etc. sauber durch. Lediglich das Durchführen der Kalibrierung fährt die Probe 3x raus und bringt dann bringt dann den "STOP called because of BLTouch error - restart with M999"

MaGe-1, 2021-07-24:
Schau in meinen anderen request. Du musst nur die eine Zeile löschen. Dann läuft es.

MaGe-1, 2021-07-24:
Schau in meinen anderen request. Du musst nur die eine Zeile löschen. Dann geht es.

knutwurst, 2021-07-30:
Ist der Pin denn korrekt geändert? Welcher Sourcecode wurde verwendet? Welche Zeile geändert? Funktioniert der Sensor, wenn man ihn direkt auf das Board steckt?

Klingt für mich nach keinem Firmware-Bug, sondern einem Hardware-Problem.

Datafreak, 2021-07-30:
Hi @knutwurst,

es geht um folgendes Kommentar:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/158#issuecomment-880194250

Ich hatte das gleiche Problem. Und nachdem ich die Zeile "if ENABLED(KNUTWURST_BLTOUCH, KNUTWURST_CHIRON)" und das passende "#endif" gelöscht habe, funktionierte der BL-Touch bei mir fehlerfrei.

Gruß,
Datafreak

knutwurst, 2021-07-30:
> Ich hatte das gleiche Problem. Und nachdem ich die Zeile "if ENABLED(KNUTWURST_BLTOUCH, KNUTWURST_CHIRON)" und das passende "#endif" gelöscht habe, funktionierte der BL-Touch bei mir fehlerfrei.

Wie gesagt - diese Stelle im Code gibt es nicht mehr. Der Pin wird über die pins_RAMPS.h geändert. 

So ist es in der 1.2.0 Final.

Das mit der Ultrabase sollte nicht so sein. Hast du die Defaults geladen?

Datafreak, 2021-07-30:
Das mit der "pins_RAMPS.h" ist irgendwie an mir vorbei gegangen. Warscheinlich an den anderen auch.
Daher die Verwirrung bei den Personen, welche den PIN von 2 auf 21 ändern.

Das mit der Ultrabase  muss ich noch genauer untersuchen. Vermutlich ein PEBKAC.

closed 2021-09-04 by github-actions[bot]