knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #176: Mega to mega S upgraded Firmware

Cryptec, 2021-07-26:
In the Releases and the Wiki is clear which Firmware to use for all "normal" Printers. Mine is a classic i3 Mega (1z-endstop) Trigorilla V1 with the official Mega to Mega S upgrade kit. (Titan extruder and Mega S firmware). 

A Note for the correct software to use would be useful in the Wiki.

knutwurst, 2021-07-26:
Hi :) 

> A Note for the correct software to use would be useful in the Wiki.
That's not possible, because there is no dedicated firmware for this kind of modification.

If you upgraded your extruder but not the endstops (so you still have only one) you should use the 1G Firmware and then use pronterface to edit the acceleration/feedrate and step values for your extruder. 

The following commands have to be entered after loading the firmware defaults:

```
M92 E400
M203 E30
M500 
```

I hope this works for you. It's the normal procedure if someone changes the extruder of a "normal" printer.

Best,
Olli

Cryptec, 2021-07-26:
wow, that was a quick answer! Thanks for this! I´ll try it tomorrow :) 

Best, 
Martin

closed 2021-08-31 by github-actions[bot]