knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #178: [BUG] (M211 S0 should be part of the Bltouch Z offset setup instructions)

bassamanator, 2021-07-29:
### Bug Description

In the configuration section of the Bltouch installation wiki, it is possible to run into the situation where you cannot lower the nozzle any further so that the paper becomes difficult to move (even though you can continue to send the G1 Z-1 command, for example). In such cases, the `M211 S0` (disable software endstops) should be included in the instructions so that the nozzle can be lowered further. I think this step is necessary to ascertain an accurate Z offset. Later the software endstops should be reanabled with `M211 S1`.

knutwurst, 2021-07-29:
While this is a good idea, it does not apply to the Anycubic if you still have hardware endstops.

After homing, the soft end stops are measured using the limit switches. If you switch this off, you can move the nozzle deeper onto the bed than the hardware allows. So you can destroy the end stops.

Instead, it makes more sense to level at Z = 0 when setting the offset. So there is no distance from the nozzle to the bed, as the calculation starts from here.

I think it is more critical to deactivate the soft end stops than to level correctly.

bassamanator, 2021-07-29:
Yes, you're right, this is potentially dangerous and should be avoided if at all possible. 

Just in passing, it was unclear to me where I should attach the 3dtouch servo cable on my Mega S 0.0.3. I think something like this might help:

[image]

Thanks for this firmware! Been using it since 1.1.9, currently on `MEGA_S_DGUS_TMC_BLT_10_v1.2.0.hex`.

closed 2021-07-29 by bassamanator