knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #180: [BUG] x axis stop switch ignored on chiron

mlewis-everley, 2021-08-01:
### Bug Description

Just tried to install this on my first generation Chiron (1.0 display and Trigorrila 1.0 board). After install, selecting `Load FW defaults` and `Save EEPROM` I try to Home X and the print head moves until the switch is depressed but then tries to keep moving (seeming to ignore the switch).

Home Z and Home Y work as expected (stop moving once the switch is depressed/IR sensor is tripped)

### Steps to Reproduce

Install 1.2.0 firmware on first generation Chiron
From the menu select Tools > Home > Home X
Allow print head to Home

**Expected behavior:** Print head to stop moving once X Axis stop switch is depressed

**Actual behavior:** Print head doesnt stop moving once X Axis stop switch is depressed

mlewis-everley, 2021-08-03:
Is there anything I can do to help with this issue? I have platform IO and auto build marlin installed. Happy to try stuff and see if it helps?

mlewis-everley, 2021-08-04:
Not sure if anyone will read this, but I notice that stock Marlin now has support for the Chiron, looking at it's config file I noticed there is an X_MAX_PIN setting that does not appear to be in this config:

https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/AnyCubic/Chiron/Configuration.h#L147

Just wondering if this might be related?

knutwurst, 2021-08-04:
Sorry mate, I'm here! 
It's a bit strange that the endstop doesn't work on your printer. Everyone who has tested the firmware so far has been able to confirm that everything works.

But I'll dig into it!

Maybe you can try the original marlin and find out if it works for you? 
The X_MAX is defined in the [pins_RAMPS_CHIRON.h](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/b3f4dcb2d1bcf2f6f3148945852d49edc2d35106/Marlin/src/pins/ramps/pins_RAMPS_CHIRON.h#L92) so this can't be the issue.

Best,
Olli

mlewis-everley, 2021-08-04:
Thanks for the reply. I have just compiled the original marlin to try it, but currently my hotend is in a lot of pieces (I thought I would take the down time to upgrade to a v6 all metal)!

I will try and find some time to see if I can test it properly.

Also, thanks for pointing out that file. Been trying to reverse engineer your setup, but as I am only just getting my head around Marlin it's been taking some time :-)

mlewis-everley, 2021-08-05:
OK, So I compiled and ran vanilla Marlin 2.0.x and same thing happened (x axis crashes and ignores the sensor).

No idea why, it was working fine with the stock firmware :-(

knutwurst, 2021-08-05:
Can you flash back to stock to ensure that the issue only applies on custom firmwares?
If so, we need to figure out, where and if the endstop is connected to a different pin.

It might be a hardware issue and a really bad timing :/

mlewis-everley, 2021-08-05:
OH MY GOD I AM SUCH A KNOB!

Installed stock firmware and the the error still persists. Disconnected and cleaned the end stop and then just tested pressing it while the x-axis is moving and it worked fine, but it still didn't work when attached.

Further investigation showed the cable running through the cable chain was too tight (it must have moved when I moved the printer to upgrade the firmware). I pulled the cable out and looser a bit and now it works fine!

I am closing this issue, I have upgraded the firmware and all seems to be working fine now.

closed 2021-08-05 by mlewis-everley