knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #184: [FR] Bump Marlin to at least 2.0.6.1 for better arc support (and more benefits)

gloomytree, 2021-08-14:
### Description

Hi, I am not sure if this is a bug or not, so I file it as a feature request. I've been printing with arc welder (done in cura) using an Octopi and experienced a lot of stutter on detailed areas. Since then I've been digging and found resources recommending Marlin 2.0.6 when using arc welder. And indeed, the Marlin release page states "Fix G2/G3 Arcs segment bugs". 

I think this repository uses Marlin 2.0.5.4, at least that is what version.h says. Are there any plans to bump Marlin to a newer version? I know this is a major effort, but it might be worth it for better arc support. There are additional benefits, with bugfixes for the BLTouch, TMC, power loss resume (I know this was an issue before), and even ExtUI support for the I3 Mega. There are even more Anycubic/Trigorilla & LCD related fixes in later Marlin versions. 

#### Additional Information
Quote from here https://plugins.octoprint.org/plugins/arc_welder/ : 

> Marlin has supported arc commands for a long time. However, starting with version 2.0.6 arc support has been greatly enhanced. I recommend you upgrade to at least this version before using Arc Welder because your experience will be much better.

Marlin release 2.0.6: https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.6

knutwurst, 2021-08-16:
Hi @gloomytree ;)
My current source code is based on 2.0.5.4, that's right. However, I merged various bug fixes from 2.0.7 and 2.0.8 upstream.
This is actually one of the big problems: Due to my own implementation of some functions and especially the implementation of the Anycubic Display, it is extremely difficult to transfer this to another original Marlin version.

I recently built a 2.0.9.1 version for the GEEETECH A10, which worked very well and I also thought about doing this for the Anycubic printer.

However, the effort is very big and actually bears no relation to the benefit. Mainly because since 2.0.6 the path to ExtUI has been taken, which is completely incompatible with my implementation. (The special menu would be really shitty)

As a quick win, I would like to offer you to take a look at the G2/G3 implementation in the 2.0.9 version and to transfer it to my Anycubic version. This is certainly the simplest solution with the greatest added value.

Olli

Jack77777777, 2021-08-16:
It would ve great to habe G2/G3 for the i3 Mega S.

gloomytree, 2021-08-19:
I was afraid it would be this complicated. This is a complete redesign for your codebase, isn't it? It would be nice ton backport the 2.0.9 implementations! 

I am wondering how this FW will continue in the next years if Marlins version can't be bumped. It would be a shame to have all this work go stale!

ashleysommer, 2021-08-20:
@knutwurst 
Would be great if you could backport the new G2/G3 features to your project!

ashleysommer, 2021-08-20:
I've started doing the backport of G2/G3 features. I just noticed, @knutwurst already has one of the ARC stuttering fixes applied already: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/c14f93323c2bc796f90b00d7b75493d6ce05c8a6

knutwurst, 2021-08-20:
1.3.0-beta.1 should be available within the next hour.

gloomytree, 2021-08-20:
wow that was fast.

gloomytree, 2021-08-20:
I've tested this now. I wrote a small script that generates arc based stress-test gcode with increasing resolution. I basically move in 180° arcs with a 5mm radius and break the arc into more and more individual parts to increase the difficulty. 

With the old fw version, the printer war OK with 3 segments and stuttering extremely with 10 segments on a single arc. 

The new fw could manage 30 segments OK-ish (maybe a bit slower) and started stuttering with 50 segments. 40 segments were noticeably slower, but I think this might work without creating print artifacts, so good work on the buffer underrun protection through reduced print speed. 

I'd call it a huge success! Are you interested in the testing gcode and or script? I've added it to this post. Beware, it's a VERY crude python script, I'm no programmer and this was meant to generate the file only once. It can be configured a bit, tough. 

[Arc Tester.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7024098/Arc.Tester.zip)

ashleysommer, 2021-08-20:
Thanks for your testing Gloomytree.
I did another test print with ARC Welder. I printed one of [these bearings](https://www.thingiverse.com/thing:4821342) after processing with ARC welder. It was stuttering pretty badly. I think I need to check the welder plugin settings.

gloomytree, 2021-08-21:
Based on my test, the minimal arc length should be greater than 0.5mm (5mm radius, 180° arc, 30 segments = 0.5mm arc length). Feedrate was 3000, so if you are going faster, you will need less resolution. Maybe start with 1mm?

knutwurst, 2021-08-21:
Reopened the issue because fine-tuning of the various settings is obviously necessary. :)

Jack77777777, 2021-08-22:
Hi Guys,

What is the difference of 1.2.0 and 1.3.0b1 regarding ArcWelder Support? I can not see any quality change for my test print with default ArcWelder settings.

gloomytree, 2021-08-22:
Knutwurst backported a few bugfixes regarding G2 and G3 from Marlin 2.0.9 which reduce stuttering and increase the performance. Maybe you had models that had no complicated geometric that leads to a high amount of arc commands in a short time... Try printing the gears from the comment above, that is a pretty good torture test. Or try my designated torture test gcode ;-)

Jack77777777, 2021-08-22:
Ok. I will try the teat. Any recommentations for AW sertings and habe them to be changed for every print ?

ashleysommer, 2021-08-27:
Worth noting, I did find a bug in the current implementation: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/pull/191
Will need retesting after that is merged.

knutwurst, 2021-08-28:
The fix is merged :) 
I'll trigger a build later. We have friends over at the weekend and in general I haven't gotten around to doing anything on the firmware lately. :/ 

Thank you both!

Esquilo007, 2021-08-28:
Thank you all for that great feature,  I will follow this thread here

PR #186 (2021-08-20): backport G2/G3 changes from upstream to fix ARC issues
Fixes #184
Testing now with ARC Welder, on Anycubic MEGA S, Old TFT, no BLT.
closed 2021-10-03 by github-actions[bot]