knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #188: [BUG] (short description)

HaGueRe, 2021-08-22:
### Bug Description

Printer: Anycubic Mega-S with TMC2208 FYSETC,
Firmware: MEGA_S_TMC_v1.3.0-beta.1.hex

After first Activation the Board-Fan runs permanently on my Mega-S.

### Steps to Reproduce
Run any Axis

**Expected behavior:**
Board-Fan stops after 3 minutes of inactivity.

**Actual behavior:**
Fan keeps running.

Best regards

knutwurst, 2021-08-22:
Hi,
the fan stops if all steppers shut down. The automatic idle timeout to release all stepper motors was disabled for testing purpopse. 

It's fixed in https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/1ecca1334722183371b5de1aae61e9f7a86a2d06 and will be available in the upcoming release.

In the meantime, use the GCODE M18 or M84 to disable all steppers after a print. The fan will be shut down 1 minute after all steppers are disabled.

closed 2021-08-22 by knutwurst

HaGueRe, 2021-08-22:
This is good to know.
Thanks for your quick reply.

Best regards