knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #189: [BUG] BLTouch fährt nach dem Einrichten ins Druckbett

Simforyou, 2021-08-23:
### Bug Description

Ich habe meinen Anycubic I3 Mega S auf X-Carriage MK4 mit BLTouch umgebaut. 
Dazu habe ich die Anleitung von Knutwurst genutzt und habe die Firmware mit S, für 1.0 Board installiert. 
Der BLTouch ist ein originaler. Ich habe ihn an B5 + B6 angeschlossen und die Servoleitung an D11 angeschlossen. 
Ich bin nach der Anleitung vorgegangen. Die Probe macht den Selbsttest erfolgreich, die blaue LED ist an. 
Wenn ich nun das Z-Offset ermitteln möchte, und ich beim zweiten fahren auf Papier an den Endschaltern der Z Achse angekommen bin, kann ich da nicht drüber hinaus fahren mir Pronterface. 
Ich habe dann zum Testen ein Offset von -1,5 angegeben und das Leveling gestartet. 
Er hat auf die Endschalter gehomed, ist dann 10mm hoch gefahren , die Position x - y 40mm angefahren,  hat die Probe ausgefahren, rot erlischt, und fährt abwärts. so lange bis die Düse aufliegt, das Bett am Ende ist und der Riemen überspringt. 
Das macht er 3 Sekunden und gibt dann in Pronterface ein "Probing failed" aus. 

Liegt das Problem an dem Z-Offset ? Ich habe zum testen Z 70mm hoch gefahren und dann mit dem Finger die Probe betätigt. Er zieht sie nicht ein.  
Per Pronterface kann ich ein/ausfahren.

Kann mir jemand helfen ? 

MFG: Christian

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2021-08-29:
Hallo Christian,
da dies kein Firmware-Bug ist, möchte ich dich bitten, die Frage in der [offiziellen Facebook-Gruppe](https://www.facebook.com/groups/3094090037303577) zu stellen, da es Viele gibt, die solche Probleme beim Einbau des Sensors haben.

Kurz und knapp: Es liegt zu 50% an falscher Verkabelung, 49% am falschen Z-Offset und zu 1% daran, dass eine "Ausschuss"-Hotendplatine von Anycubic verbaut wurde, auf der sich noch eine Reihe von SMD Widerständen befindet. Da müssen dann die Widerstände runter, da der Sensor sonst nicht auslöst. Der Fall ist aber eher, dass er dann "immer" auslöst.

Olli

Simforyou, 2021-08-29:
Moin Olli.Ich hab den Fehler gefunden. Der Bltouch ist defekt. Habe einen anderen eingebaut, der ging auf Anhieb.Mfg Christian

closed 2021-09-22 by knutwurst