knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #190: [BUG] Chiron "randomly" ignores z-offset set from probe/leveling menu

mlewis-everley, 2021-08-24:
### Bug Description

So I have the bed nicely tram to the frame/nozzle, using the level probe and then tweaking using the advanced menu. But every now and then the printer seems to forget this setting and starts printing ~10 - 15mm too height.

I cancel the print and then go to advanced settings, decrease the height and then increase it again to it's previous value and it works fine for anywhere from 1 - 6 prints, then I have to do it again.

Not sure why this is happening, everything else works great!

### Steps to Reproduce

1. Install latest firmware, load firmware defaults save to EEPROM
2. From the touch screen go to Tools > Level > Probe
3. Probe bed
4. From touchscreen go to Tools > Level > Advanced settings
5. Manually adjust settings as needed
6. Run some prints sooner or later the first layer will print seemingly ignoring the z-offset.

**Expected behavior:** [What you expect to happen]

Z-Offset set under Tools > Level > Print should be used for every print.

**Actual behavior:** [What actually happens]

Z-Offset isn't always used and has to be re-set/saved from the LCD.

#### Additional Information

This is a very annoying issue as it is highly intermittent. It is really difficult to try to detrmine a potential cause.

mlewis-everley, 2021-08-28:
OK, I have been trying to keep an eye on when this happens, I *think* it is whenever the printer restarts (meaning it is physically switched on or connected to via octoprint).

As I haven't re-connected or shut it down now for a few days, it seems to be running pretty stable, but last time I had to re-connect it lost the z-offset settings again.

Do the manual level settings have to be saved via an M500?

knutwurst, 2021-08-28:
> Do the manual level settings have to be saved via an M500?
Yes, if you do a mesh leveling manually or edit the z offset, you need to store it via "SAVE EEPROM" or M500.

mlewis-everley, 2021-08-28:
Thanks for the response @knutwurst . So I did an M500 and restarted the printer.

Under Tools > Level > Advanced Settings the correct value is displayed. But, if I try and print then it seems to be ignored.

As I said in the original report, if I change this setting and then tap "OK", then change it back again, the value is recognised.

So M500 doesn't seem to work. It is almost like the firmware is ignoring the z-offset set this way until you update it again via the LCD?

knutwurst, 2021-08-28:
I'll take a look at it, but so far I don't know anyone who has had this problem.

knutwurst, 2021-09-17:
After some investigation I can say: I have no idea.
I cannot reproduce the problem. :(

Maybe you have a faulty mainboard? If it were a firmware bug, it would have to occur with many more users.

mlewis-everley, 2021-09-17:
Very odd, I would do some more testing, but the umbilical cable between the body and the extruder has breaks in multiple places and so the machine is dead while I wait for a replacement.

To be honest, I think I am going to get an MKS Robin board and touchscreen and just replace the irritating wiring with something a bit more robust. If you are unable to replicate then it could be something weird going on with this printer, these dodgy cables have caused a lot of odd issues.

closed 2021-09-17 by mlewis-everley

scharrrfi, 2021-10-09:
Hi Knutwurst,

erstmal danke für die tolle Firmware und die viele Arbeit. Die Möglichkeit bei deiner Firmware nach dem Autolevel mit einem Stück Papier zu kalibrieren, finde ich super. Mit der Stock-Firmware ist das ja nur mit einem Druck und try und error möglich.

Ich kann aber das oben genannte Problem bestätigen. Das komplette Autoleveling-System des Chiron ist bei deiner Firmware irgendwie buggy.

Ich versuche kurz zu beschreiben, welches Verhalten auftritt.
Level Probe funktioniert soweit und es werden Werte generiert und im "Tools > Level > Advanced settings"-Menü angezeigt.  Das anschließende Feintuning hat bei mir nur einmal funktioniert. Die "kalibrierten" Werte waren aber völlig obsolet, nachdem ich dann einmal über das Menü nach Home gefahren bin. Danach waren alle Z-Offset-Punkte viel zu tief und er wollte sogar seitlich mit der Nozzle ins Bett fahren...

Dann habe ich nochmal neu versucht einen Probe durchzuführen (auch nachdem ich vorher über das Special Menü "Reset Level Grid" durchgeführt habe). Danach wurde das Z-Offset weiterhin viel zu tief eingestellt. Einstellungsänderungen im Advanced settings"-Menü hatten seitdem keinerlei Einfluss auf den Offset.

Zudem war das Advanced settings"-Menü nun ein paar mal nicht aufrufbar (passierte nichts). Hier hilft nur ein Reboot des Druckers. 

Irgendeine Idee oder sollten Chiron Nutzer lieber wieder zurück zur Stock-Firmware?