knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #192: Its not a bug but its an question that doesn't exist in faq or wiki

TeleportYT, 2021-08-29:
How after updating to marlin the i3 mega s can work in much more place if still the ultrabase holded by the metal holders?
Wont it collide with the nozzle?

knutwurst, 2021-08-29:
Short answer: Yes

Long Answer: it depends... The fact is that you very rarely print right to the edge where the brackets are. However, I often print objects that are simply wider than 210 mm.
There are also many users who have replaced their printing plate.

So of course you are right, but if you really want to print an object that is larger than 210x210 in X and Y, you will first see it in the slicer. And if the firmware didn't allow printing that large, the nozzle just wouldn't go any further over the point and the object would be deformed.

If you really have to use 100% of the installation space, you can also remove the (useless) brackets.

Yes, a hint on the wiki might help, but I was hoping for common sense :D

TeleportYT, 2021-08-30:
oh , okay thx :)

closed 2021-08-30 by TeleportYT