knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #193: What happened after 2-3 times using - autoleveling (BL-Touch) didn‘t start again

recaremo, 2021-09-09:
I used your firmware 1.3.beta.2 with original BL-Touch. Installing and leveling worked fine - autoleveling too.

Today I recognised a print issue and wan‘t to start autoleveling again. 
1. Start printer: BL-Touch self-test working
2. Start leveling: BL-Touch open-close - like selftest.
3. Printer goes to test position 1
4. BL-Touch close and printer beep.
5. Nothing happened.

knutwurst, 2021-09-09:
Connect a PC with pronterface and read out the error message.

If it shows 999 BL Touch Error, your wiring is wrong or the sensor defective.

If it shows probing failed, your Z offset is not within tolerance and the nozzle had to move too far. Try 4 corner easy leveling and move the nozzle down until it slightly touches the bed on all 4 corners.

recaremo, 2021-09-09:
Thanks I’ll check it.

recaremo, 2021-09-09:
Recv: Error:!! STOP called because of BLTouch error - restart with M999
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N8 M112*41
Send: N9 M104 T0 S0*40
Send: N10 M140 S0*84
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

Well that's not your problem. Do know why an original BLT is defekt after 2-3 times using?

recaremo, 2021-09-09:
and calibration process worked fine

recaremo, 2021-09-15:
Sorry, was my mistake. Changing BLTouch solved all problems. Please close this issue.

closed 2021-09-22 by knutwurst