knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #194: [FR] Expose additional UART pins for ESP3D

PapeCoding, 2021-09-09:
### Description

Out on GitHub there is another repository which enables a lot of printers to have wifi access. It is called [ESP3D](https://github.com/luc-github/ESP3D). They even have a [tutorial ](https://github.com/luc-github/ESP3D/wiki/Trigorilla) on how to get it working with the Trigorilla board. Unfortunately, this requires hijacking the UART pins from the USB port and soldering directly to the main board. Is it possible to somehow expose some already pre-soldered header pins on the board as UART pins to use with the ESP3D?

### Feature Workflow

1. Flash ESP3D board
2. Solder Jumper wires to the board
3. Put them somewhere on the board

#### Additional Information

* Especially interesting would be to still maintain the USB functionality while ESP3D is running.

knutwurst, 2021-09-11:
Normally I would say: If you already have an ESP8266, you can swap the board for something fast and use RepRap firmware.

In this case I find it interesting that you can use an ESP01! That's really cool and I didn't know it before.

A second serial interface is theoretically possible directly. I just have to see which pins can be used for this.

Anyway, you made me curious ;)

knutwurst, 2021-09-12:
@PapeCoding , I added an additional port which shares the command queue so it should work with USB simultaneously. The Pins are D16/D17 on the EXP1 header which are unused in most anycubic printers.

knutwurst, 2021-09-12:
Hi.. I did some testing and..... it works!
The feature will be included in the next release.

[image]

[image]

[image]

Esquilo007, 2021-09-12:
Cool feature 👍 I will also try this and give feedback

PapeCoding, 2021-09-14:
Oh, wow, thank you @knutwurst!

I did not expect it to be integrated that fast! I am on holiday and will check it out next time I am on the printer :)

knutwurst, 2021-09-14:
I would say: It's done :) -> https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/WLAN-Installation-(deutsch)

Currently only in german... but for now, I'm happy with the result.

closed 2021-09-17 by knutwurst