knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #195: [BUG] Bltouch probing failed

wassereimer86, 2021-09-13:
### Bug Description

Vorab:
Ich habe das Wiki gelesen und versucht das Problem zu lösen. Leider weiß ich nicht weiter und versuche daher hier mein Glück. Vielen dank vorab für die FW und die Arbeit!

Problem:
Bltouch Leveling funktioniert nicht. Im Video das Verhalten: https://youtu.be/asEZF-AKxls

Einrichtung nach Anleitung hat funktioniert. Sensor reagiert auf die Befehle in der Anleitung und fährt aus und ein. Bett habe ich manuell so gut wie möglich eingestellt. Abstand von Bltouch und Nozzle Spitze ist im Rahmen der vorgegebenen Parameter (zwischen 2,3 und 4,3mm - ich bin bei 3,1 mm). FW installiert habe ich MEGA_S_BLT_10_v1.2.0. Habe zwar neue Treiber aber die Anschlüsse manuell gedreht da der vom Extruder noch der alte ist. Ich müsste das Trigorilla 0.0.2 Mainboard haben. Wobei ich am Mainboard keine Versionsnummer finden konnte und es nicht mit denen aus der Anleitung übereinstimmt. Der Bltouch wird aber ja scheinbar richtig gesteuert. Pronterface gibt nur "probing failed" aus.

**Expected behavior:**

Kein "probig failed" und autoleveling funktioniert

**Actual behavior:**

Siehe Bug Description.

knutwurst, 2021-09-13:
Hallo @wassereimer86 ,
Im Video ist ein schnelles blaues Blinken zu sehen - das heißt schon, dass ein Fehler vorliegt. Vermutlich sind die Sensor-Pins (schwarz/weiß) vertauscht, oder die Sensor-Spitze lag auf dem Bett auf, als du den Drucker gestartet hast. Auf jeden Fall funktioniert der Sensor so nicht.

Kannst du die Spitze mit M280 S60 oder S160 bewegen?

wassereimer86, 2021-09-13:
Hey @knutwurst ,

danke für die schnelle Antwort! Die spitze lag beim Start nicht auf. So sieht es aus wenn das Gerät startet. Für mein Auge leuchtet der Bltouch durchgehend blau und rot. Hier im Video zu sehen:  https://youtu.be/Bafaq-grdc4

Auf M280 S60 oder S160 erhalte ich keine Reaktion. Mit M280 P0 S10 aus der Anleitung fährt er aber aus und, wenn er ausgelöst hat, mit M280 P0 S160 wieder ein.

wassereimer86, 2021-09-13:
Es war mein Fehler... Schwarz/Weiß war nicht vertauscht sondern an den falschen Kabeln. Ich hab das direkt am Kabelbaum abgegriffen und, anhand der Grafik im Vergleich zum eingebauten Zustand, falsch gespiegelt. Zum Glück hat der Sensor, so wie es scheint, überlebt. Entschuldige das ich deine Zeit geraubt habe.

Die Frage im "Edit" meines letzten Beitrages würde mich aber noch interessieren. :)

knutwurst, 2021-09-14:
Guten Morgen @wassereimer86 ... und vielen Dank ;) Wäre nicht nötig gewesen, da ich meine Freizeit in das Projekt stecke und - falls ich mal keine Lust habe - einfach nicht antworte :D Also alles gut :) 

Puh... Glück gehabt. Das hätte den Sensor auch zerreißen können, wobei ich nicht weiß, auf welchen Pins du warst. Vermutlich Heizung oder Lüfter. Da beides nicht eingeschaltet war, hattest du Glück. Aber schön, dass es jetzt funktioniert! Das freut mich :) 

Zu deiner Frage im Edit: Die Anleitung ist noch vergleichsweise alt und nicht sehr ausführlich. Ich wollte sie sowieso mal überarbeiten. Ja, Z=0 bestimmt man mit aufliegender Düse. Also wirklich knapp, weil wir ja den Abstand vom Bett zum ausgelösten Sensor messen wollen. Deswegen fahren wir auch erst den Sensor und dann die Düse an dieselbe Stelle auf dem Bett... einfach um Fehler auszuschließen.

Allerdings kannst du seit der 1.3.0 Beta 2 den Z-Offset einfach im Menü einstellen und bei Bedarf nachjustieren. Also klar ... das ist kein Babystepping, weil du ja während des Drucks nicht in das Menü kommst. Aber du kannst von dort aus einfach mal probieren und neu Leveln lassen.

In die 1.3.0 kommt auf offizieller Support für 5-Euro-WLAN. Also wäre es ohnehin super, wenn du die letzte Beta testen würdest, damit ich vorzeitig auf etwaige Fehler stoße. Die Version hat ja auch ein paar andere neue Features wie Meatpack und G2/G3 Support.

lg
Olli

wassereimer86, 2021-09-14:
Hey @knutwurst ,

gerne. 🙂

Ich bin zum Glück nur auf A5 und A6 gelandet. Sonst wäre vermutlich was hinüber gewesen. 😅

Ok, gut zu Wissen. Vielen Dank! Dann hab ich es ja später zum Schluss hin irgendwann instinktiv richtig gemacht die Nozzle fast auf das Bett zu legen. 👍

1.3.0 Beta 2 hab ich direkt installiert gehabt, nachdem ich gelesen hatte, dass das Z Offset damit nun über das Menü angepasst werden kann. 😁 Ich bin jedoch noch Anfänger und vermutlich nicht wirklich hilfreich bei Beta-Tests. 😄 Falls ich aber was testen kann, sag bescheid. 🙂

Viele Grüße
Jens

closed 2021-09-16 by wassereimer86