knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #196: [FR] G34 - Z Steppers Auto-Alignment

melrhombus, 2021-09-13:
### Description
Enable G34 - Z Steppers Auto-Alignment

### Feature Workflow
1. Send G34
2. Printer probes bed to align z motors
3. Reports accuracy

#### Additional Information
The Mega-S has dual z-motors but the Z_STEPPER_AUTO_ALIGN is commented out and the second Z motor is defined as E1 instead of Z2. More than anything I'm curious why this is defined as such and if it's possible to switch things around to allow the G34 in a BLTOUCH enabled release

knutwurst, 2021-09-22:
I had implemented this feature before, but then expanded it again. It works satisfactorily with a printer with only one Z end stop, but with two Z motors aligning the X axis takes so long that leveling is no fun. Also, you are forced to do this every time you do G28.

For these reasons, I returned to the two Z limit switches and only use the BLTouch to compensate for unevenness.

Sorry.

closed 2021-09-22 by knutwurst

kb-elmo, 2021-09-28:
Is there any chance that this could be implemented again in a separate version?

Having this feature would be really nice to have for many people I think.

It also seems that there is a pullrequest for this open but kinda abandoned?
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/pull/61

And just a sidenote:
If the auto align takes too long it's usually an indicator that the amplification factor was just set too low.

kb-elmo, 2021-09-28:
Ignore my last comment 😄 

I decided to dive into Marlin building myself and found that it actually wasn't too hard to implement this myself.

I have G34 working now with both Z-endstops removed and the BLtouch acting as the Z-min.
The BLtouch also uses the usual pins and not the Z-endstop pins making it easier to use if you already installed the BLtouch according to the guide.
With both Z-steppers somewhat out of sync it now synchronizes them back to 0.02mm tolerance in 3-4 iterations.

Should I create a pullrequest for this so that other people can make use of it as well?