knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #198: Meatpack Support not working on i3 Mega S

Ekodas, 2021-09-17:
I am currently running on a custom configured 1.3.0-Beta-2 and I have been unable to get Meatpack to work.  (Had the same issue with 1.3.0-beta-1).  I have enabled Meatpack in the firmware and installed the Meatpack addon for Octoprint.  However, whenever I turn on Meatpack and restart my Octoprint server, nothing works correctly and I get repeated unknown command errors.  Is there something that I am missing, or is Meatpack not compatible with all Anycubic motherboards/printers?

I have an Anycubic i3 Mega S with the Trigorilla 1.0 board.

Thanks!

knutwurst, 2021-09-17:
Hi @Ekodas,
Unfortunately, I can't tell you what's wrong. I only tested the feature briefly, but otherwise do not use Octoprint productively.

All users who have been using Meatpack + Octprint up to now have had no problems. In any case, I don't know of any problems.

Maybe @ashleysommer can give you a tip because she's the one who implemented the feature. (See: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/pull/185)

ashleysommer, 2021-09-17:
@knutwurst 
> because she's the one who implemented the feature

\* He

@Ekodas 
I have the same model Mega S with the same Trigorilla v1.0 board, and it works fine for me. I've using 1.3.0-Beta1, (I haven't tried 1.3.0-beta2).
There should be nothing you need to do in the firmware to turn it on, it is compiled in and already enabled. Installing the plugin for OctoPrint is the only requirement.
I'm sorry, I don't know how I can help. It would be good to ask someone else with the same printer and same firmware if it works for them.

knutwurst, 2021-09-17:
@ashleysommer 
>  * He

I'm so so sorry! The name Ashley sounded so feminine ... I'm embarrassed now. Sorry >.<

Also.. the Beta 3 is out now with some minor fixes (also in the meatpack implementation regadring more than one serial interface) and WIFI Support.

Ekodas, 2021-09-19:
So I think I figured out the issue.  When I flash the firmware and then enable MeatPack in Octoprint, it forces a reboot of the OctoPrint server.  Once OctoPrint comes back up, I'm receiving the error messages.  At that point, if I force OctoPrint to disconnect from the printer and then re-connect, everything then seems to be working properly with MeatPack enabled and connected.  Hope this helps if anyone else has experienced an issue with "unknown commands" with MeatPack enabled.

closed 2021-09-19 by Ekodas