knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #199: [BUG] Wiki G-Code / Update Anleitung

wassereimer86, 2021-09-18:
Hey,

ich habe mich mal mit dem GCode auseinandergesetzt und denke das deiner aus dem Wiki einiges doppelt angibt.

> M140 S{material_bed_temperature}           ; Start heating the bed
> G4 S60                                     ; wait 1 minute
> M104 S{material_print_temperature}         ; start heating the hot end
> M190 S{material_bed_temperature}           ; wait for bed
> M109 S{material_print_temperature}         ; wait for hotend

Für mich heißt das:
- Starte aufheizen des Bettes und mach weiter
- 1 Minute warten
- Hot End aufheizen und weiter machen
- Bett aufheizen und warten bis Bett heiß ist
- Hot End aufheizen und warten bis Hot End heiß ist

Ich habe das bei mir jetzt geändert zu:
> M190 S{material_bed_temperature}   ; wait for bed
> M109 S{material_print_temperature} ; wait for hotend

So legt er los sobald er die Temperatur erreicht hat und gibt keine doppelten Befehle. Oder übersehe ich hier etwas? :)

Zweites Thema: Eine Update Anleitung wäre schön. Ich scheue mich etwas die Beta zu verlassen, da ich den Extruder kalibriert habe etc. Das müsste ich dann ja alles neu machen? Gibt es dafür ein Workaround oder eine Lösung?

Viele Grüße

HaGueRe, 2021-09-18:
Falls ein "Load FW Defaults" erforderlich ist, gebe ich vor dem Aufspielen einer neuen FW am Terminal "M503" ein und speichere die Ausgabe in einer Textdatei. Nun habe ich alle relevanten Daten gesichert und entferne aus dem Text alles außer den Zeilen, die 

> M92
> G29
> M301
> M304
> M900

enthalten.

Aus den verbliebenen Zeilen lösche ich zum Abschluß noch "Recv: echo:  " heraus und setze an den Anfang der Datei noch ein "M502" und "M500". Das ist aber optional und vermutlich nicht unbedingt nötig. Wichtig ist nur, die Textdatei mit einem "M500" in der letzten Zeile abzuschließen.

Nun muß man nur noch die Endung "*.txt" in "*.gcode" umbenennen und die Datei wie gewohnt auf den Drucker kopieren und drucken. Alle von dir erarbeiteten Werte sind jetzt wieder im EEPROM und alles sollte wieder wie gewohnt funktionieren.

Noch zur Erklärung: Die gesicherten Daten betreffen

> Steps per Unit
> Mesh Bed Leveling
> PID Hotend
> PID Bed
> Linear Advance

Auf diese Weise ist es mir möglich, zwischen verschiedenen Versionen hin- und her zu switchen, ohne jedesmal den Drucker neu  kalibrieren zu müssen.

Gruß Hans

wassereimer86, 2021-09-18:
@HaGueRe Danke dir, Hans. Für mich ist noch M851 wichtig und ich habe kein "Recv:"; aber das habe ich schon hinbekommen. :) Ein Wiki-Eintrag wäre hier bestimmt auch für andere hilfreich. :)

HaGueRe, 2021-09-19:
Danke für den Hinweis. Habe das oben mal etwas verallgemeinert bzw. ergänzt.

Gruß Hans

knutwurst, 2021-09-20:
@wassereimer86 , du hast natürlich Recht, was den Code angeht! Witzig, dass das bisher keinem anderen aufgefallen ist :D M140/M104 heizt auf, ohne auf die Temperatur zu warten, M190/M109 heizt auch auf ... und wartet.

Hab es mal angepasst. Danke!

Zu dem Thema der Einstellungen:
Ich sage das nur immer sicherheitshalber. Im Grunde muss seit der 1.1.6 nichts mehr geändert werden und man kann einfach updaten, ohne die jeweiligen Einstellungen neu einzugeben oder vorher Defaults zu laden.

Nur... erkläre das mal jemandem. Da gehe ich lieber auf Nummer sicher und lasse immer die Grundeinstellungen laden. Ebenso falls jemand von einer anderen Firmware kommt, die möglicherweise eine ältere oder identische EEPROM-Version nutzt. Dann werden falsche Werte geladen und möglicherweise der Druck komplett versaut. 

Deswegen... falls du von der 1.2.0 kommst oder schon seit der 1.1.6 weißt, was du tust bzw einen Überblick hast, dann kannst du auch einfach updaten.

wassereimer86, 2021-09-20:
@knutwurst Danke für das Feedback und die Info bzgl. den Updates! Gut zu Wissen. Und ich verstehe SEHR GUT warum du das lieber allgemein sagst. Das würde nur dazu führen, dass es mindestens 50% mehr Leute **nicht** machen würden, als es vermutlich, trotz klarer Ansage, eh schon nicht machen. :D

closed 2021-09-20 by wassereimer86