knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #200: The firmware is correct, but there are no BL.touch leveling on the special menu. Why ?

Hua-Fen, 2021-09-20:
### Bug Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2021-09-20:
Since 1.3.0 the BLTouch Feature can be found here: https://www.youtube.com/watch?v=XNBlsdDNqvQ

Hua-Fen, 2021-09-20:
> 自 1.3.0 起，此处可以找到 BLTouch 功能： https://www.youtube.com/watch?v=XNBlsdDNqvQ

Hua-Fen, 2021-09-20:
> 自 1.3.0 起，此处可以找到 BLTouch 功能： https://www.youtube.com/watch?v=XNBlsdDNqvQ

After measuring the first point, the printer returns to its starting point

Hua-Fen, 2021-09-20:
Is there a more detailed tutorial

knutwurst, 2021-09-20:
I'm sorry, but all these questions are covered in the FAQ: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/FAQ-(english)

Also there is the Wiki with all additional information: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki