knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #201: M280 p0 s10 doesn't work

phyex0, 2021-09-20:
Hi dear developer. I'm in trouble with the bltouhc. I couldn't use my Chinees clone to auto-level it. It didn't respond and I thought it's broken. So, I've updated the firmware and tried to use it again. It didn't respond again. Then I decided to buy the original bltouch.  I've started to arrange bltouch by your guide but somehow it doesn't respond to m280 command again. I've checked the connections and it seems fine. When I turned on the power Bltouch works. but now it doesn't respond to m280 commands. What to do?

phyex0, 2021-09-21:
Connecting...
start
Printer is now online.
echo:Marlin 2.0.x
Marlin-AI3M
echo: Last Updated: 2021-09-18 | Author: (knutwurst)
echo:Compiled: Sep 18 2021
echo: Free Memory: 1232  PlannerBufferBytes: 1568
echo:V76 stored settings retrieved (712 bytes; crc 24143)
echo:  G21    ; Units in mm (mm)
echo:; Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:; Steps per unit:
echo: M92 X79.21 Y79.21 Z404.04 E477.40
echo:; Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z6.00 E30.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y2000.00 Z60.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1500.00 R1500.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z0.00
echo:; Endstop adjustment:
echo:  M666 Z0.00
echo:; PID settings:
echo:  M301 P15.94 I1.17 D54.19
echo:  M304 P251.78 I49.57 D319.73
echo:; Z-Probe Offset (mm):
echo:  M851 X-2.00 Y-25.00 Z-0.40
echo:; Linear Advance:
echo:  M900 K0.00
echo:; Filament load/unload lengths:
echo:  M603 L538.00 U555.00
echo:SD init fail

This is the output when I connect my device via pronterface.

knutwurst, 2021-09-21:
Hello,
can you show me some photos of your soldering? Also I need a picture of all connections you made and one from the small pcb in the print head.

Thanks in advance!

phyex0, 2021-09-21:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

All of those connections and soldering are controlled via a multimeter. It seems all correct and works.

phyex0, 2021-09-21:
Also, it was the same before changing the Chinees clone and it was working. I just changed the black-blue-purple colored wires today and it seems working.

knutwurst, 2021-09-22:
So.. as I understand... it's working now?

phyex0, 2021-09-22:
no it was working before but somehow it's not working now

knutwurst, 2021-09-22:
Ok.. what I see is that your wiring seems to be wrong.

1. The white cable (z-min) goes to the G3 pin which is the one which points to the edge of the PCB. In your pictures it's the other way around.
2. I can not see or follow your servo cables. The color coding on the Trigorilla Board seems ok to me, but they might have changed somewhere... 

Also I assume that the sensor itself shows an error when booting up, right? Slow or fast flashes.

phyex0, 2021-09-22:
I'll check the wiring at pcb again. Color coding seems correct. I'll check it again.

There is no error at Bltouch. There is only red color and no flashing. I can't see any error log at pronterface

phyex0, 2021-09-24:
I can't use M43 to debug pins. Why?

https://marlinfw.org/docs/gcode/M043.html

knutwurst, 2021-09-24:
Because PINS_DEBUGGING ist disabled for production use. That's intended.

phyex0, 2021-09-24:
> For me it seems that white and black are switched.

I changed white and black wires but the result is the same.

phyex0, 2021-09-24:
What else I can do to solve this problem. The problem could be the PCB at the head?

knutwurst, 2021-10-19:
Hi @phyex0 ,
sorry for the late answer. If you replaced the pcb and the sensor still doesn't work, maybe it's defective? Can you try another bltouch sensor? 

You can debug the behaviour if you start the auto leveling and then short the black and white wire out so the sensor think that it's triggered.

phyex0, 2021-11-02:
https://photos.app.goo.gl/BR65FSCeq5HQkvUh7 i tested the sensors on arduino. Pin in pin out and self test all worked for the both sensors

What should I do?

knutwurst, 2021-11-02:
I would say: rewire everything. It must be a bad connection on some of your wires.

phyex0, 2021-11-02:
> I would say: rewire everything. It must be a bad connection on some of your wires.

I've already did this. I also used muter to be sure wires are not broken

knutwurst, 2021-11-02:
But your cables probably have too much contact resistance. I can't do any more either. I'm sorry.

phyex0, 2021-11-02:
Well when I directly connect sensor to motherboard it didn't work too. It could be motherboard issue?

knutwurst, 2021-11-02:
Try to connect the sensor directly to the board which is installed in the printer. You need to home first, then you can control the Sensor. Also it might be that you need the 12V.

Did you try another firmware? Please test the [1.1.6](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.1.6).

phyex0, 2021-11-05:
I'm going to try this in my free time. I'll connect the board to my PC via a type-B USB cable. Still, need 12?

knutwurst, 2021-11-05:
I would say... yes... since it might need more power from the voltage regulator on the board.

phyex0, 2021-12-05:
I sold the 3d printer lmao.

closed 2021-12-05 by phyex0