knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #202: [BUG] (short description)

miscrtnone, 2021-09-21:
### Bug Description

### Steps to Reproduce

Gestestet nur mit Version 1.3

Benutze testweise VSC + PlatformIO auf Windows nicht! Linux um die unterschiedlichen Firmwareversionen zu erstellen. 4 Stück werden nicht kompilieren mit dem Fehler avr-g++: CreateProcess: No such file or directory

**Expected behavior:** [What you expect to happen]

Ohne Änderungen sollte alle kompilieren sowohl auf Windows als auch auf Linux

**Actual behavior:** [What actually happens]

MEGA_S_DGUS_TMC_BLT_10
MEGA_S_DGUS_TMC_BLT_11

MEGA_X_DGUS_TMC__BLT_10
MEGA_X_DGUS_TMC__BLT_10

kompilieren mit einem Fehler

[image]

#### Additional Information

Testweise habe ich auf Linux (Mint) die MEGA_S_DGUS_TMC_BLT_10 kompiliert und hier hat es geklappt.
Meiner Recherche nach gibt es ein maximales Kommandoausführungslimit bei Windows, dass bei den oben genannten evtl. überschritten wird. Vielleicht hat es etwas damit zu tun, das ist zumindest meine Vermutung.
Dieses Video beschreibt es näher.
https://www.youtube.com/watch?v=6kujTCn7sj4
Die meisten Nutzer von Konsumentendrucker werden sich in der Windowswelt bewegen, so auch ich, da absolute Leadplattform im Desktopbereich. Gleichzeitig gibt es doch einige Änderungen, die man gerne selbst einpflegt in deine tolle Firmware individuell an seinen Drucker anzupassen, sodass die  fertigen .hex Dateien oft keine Lösung sind.

Ich hoffe es gibt dazu eine Lösung, z.B. Dateien herausschmeißen die nicht gebraucht werden, damit dieses Limit nicht überschritten wird (wenn meine Vermutung richtig ist)

knutwurst, 2021-09-22:
Hallo,
das Problem ist bekannt und tritt tatsächlich durch die Länge des Pfades unter Windows auf.

Ein Workaround ist, die Profile einfach in etwas kürzeres umzubenennen. Diese befinden sich in der Datei i3_mega_envs.ini. Wenn du die Namen kürzt, müsste es gehen.

Da ich mich nur unter macOS und Linux bewege, fiel mir das nie auf.. sorry. Aber jetzt noch alle Umgebungen umzubenennen würde mehr Verwirrung stiften, als es so weiterlaufen zu lassen. Außerdem kann man auch die maximale Verzeichnistiefe in Windows anpassen. : https://docs.microsoft.com/de-de/windows/win32/fileio/maximum-file-path-limitation?tabs=cmd

miscrtnone, 2021-10-02:
Hallo,
du schreibst, dass das Problem bekannt ist und vermutest, anders als ich, die Verzeichnistiefe, die jeder Nutzer selbst beheben kann. Konntest du schon unter irgend einem aktuellen Windows verifizieren, dass die Firmware (bei den problematischen Variationen) mit einem Fix oder Workaround erfolgreich kompiliert?

Die Verzeichnistiefe von Windows zu ändern mittels eines Registry-Eingriff (+Neustart) bringt leider bei mir keine Verbesserung (getestet) Version 1.3. Anbei ein Screenshot.
[image]

miscrtnone, 2021-10-06:
Ich habe mich jetzt nochmal damit beschäftigt.
Wenn ich, wie von dir als Workaround Lösung angedeutet hast (Danke hierfür), unter Windows in die i3_mega_envs.ini eine Environment für die in meinem Fall problematische Mega Konfiguration erstelle (Werte von MEGA_S_DGUS_TMC_BLT_10  übernommen), mit verkürztem Namen, plus den Namen in die platformio.ini hinzufüge bei den default_envs, dann kompiliert die Firmware.
In meinem Fall habe ich mehr oder weniger die CamelCase-Schreibweise verwendet, um die Unterstriche einzusparen. [Einen Funktionstest habe ich damit noch nicht gemacht].

Ich habe mal getestet, bei wie vielen zusätzlichen Zeichen die Firmware nicht mehr kompiliert, anbei ein Screenshot.
[image]

Ob die CamelCase-Schreibweise jetzt wirklich mehr Verwirrung stiften würden, als Windows Nutzer, die schon direkt zum Start beim Kompilieren konfrontiert werden, ... das musst du selber entscheiden. (Vielleicht ist ein sehr deutlicher Hinweis darauf auch eine pragmatische Workaroundlösung).

closed 2021-12-12 by github-actions[bot]