knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #204: [BUG] Communication timeout while printing

markecskace, 2021-09-21:
### Bug Description

<I am using MEGA_S_TMC_BLT_10_v1.3.0.hex on my Mega S with BL touch. Also octoprint. After print starts, printer randomly stops and looses connection with octoprint.
This is what I see in octoprint terminal window:
`Send: N66 G1 X105.403 Y124.354 E5.11206*85
Recv: Error:checksum mismatch, Last Line: 65
[...]
Communication timeout while printing, trying to trigger response from printer. 
Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Recv: Error:Line Number is not Last Line Number+1, Last Line: 65
`[...]>``
I have to power off/on printer. If I disconnect the connected camera on raspberry it seems to print next project ok.
I did reflash version 1.2.0. and it is working perfect with camera and octolapse. No problems with version 1.2.0. 

### Steps to Reproduce

1. [First Step]
 Start print using octoprint and connected and enabled camera.
2. [Second Step]
Wait and see if it will manage to print or not.
3. [and so on...]

**Expected behavior:** [Like version 1.2.0, but with added features of version 1.3.0. ;) ]

**Actual behavior:** [It randomly stops printing after loosing connection.]

#### Additional Information
Thank you for all your work so far!

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2021-09-22:
You are the second person who reported the issue. But it's weird, because it works on 99% of all other machines with octoprint. 

Can you give me some more information about your setup like
- Octoprint Version
- Raspberry Pi Version
- All Installed Plugins
- Comm speed

Did you try another power supply for the Pi and/or the camera?

The communication may be more "fragile" since there is a second serial interface in use. But technically it should not make any difference.

To test that, I built another version for you which disables the wifi-functionality completely. Please let me know if this fixes your issue.
[MEGA_S_TMC_BLT_10_v1.3.0-noWifi.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7210465/MEGA_S_TMC_BLT_10_v1.3.0-noWifi.hex.zip)

Best,
Olli

markecskace, 2021-09-22:
Thank you!

Will report back after some testing.

My setup:

Octoprint version 1.6.1
Raspberry Pi 4 Model B, 4GB with OFFICIAL RASPBERRY PI 4 USB-C 15.3W POWER SUPPLY, 5.1V, 3.0A

Plugins:

PrettyGCode, "url": "https://github.com/Kragrathea/OctoPrint-PrettyGCode"
GcodeEditor, "url": "https://github.com/ieatacid/OctoPrint-GcodeEditor"
Filament Manager, "url": "https://github.com/OllisGit/OctoPrint-FilamentManager" 
Exclude Region, "url": "https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin"
Camera Settings, "url": "https://github.com/The-EG/OctoPrint-CameraSettings"
Octolapse", "url: "https://github.com/FormerLurker/Octolapse" 
Calibration-Companion, "url": "https://github.com/iFrostizz/OctoPrint-CalibrationCompanion"
Bed Visualizer, "url": "https://github.com/jneilliii/OctoPrint-BedLevelVisualizer"

Baudrate: 250000

I did not try another power supply. Camera is Logitech c270 webcam plugged directly into the Pi4.

Daddycool2708, 2021-09-22:
Hello
Yes I have the same issue. But the printer is still sending the temperature of the bed and hotend. 
My setup ist a Raspberry Pi3 with octoprint 1.6.1 and the official power supply. The camera is the official Pi cam.

The installed plug ins are

Print time Genius
Filament Manager
PSU power control
Arc welder
Cura thumbs
Exclude region

knutwurst, 2021-09-22:
@Daddycool2708
[MEGA_S_v1.3.0-noWifi.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7212640/MEGA_S_v1.3.0-noWifi.hex.zip)
Here is the unmodified Mega S Version (without BLTouch and Trinamic Drivers)

knutwurst, 2021-09-23:
Thank you so much for testing! 
Would you mind to test another version linked below?
It's the same as 1.3.0 but with decreased wifi module frequency, so there is more room for the cpu to handle USB requests.

I need to find the sweet spot where both serial connections work reliable. If it's not possible at all, I'l deactivate Wifi for all USB/Octoprint users.

[MEGA_S_TMC_BLT_10_v1.3.0-115k.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7217481/MEGA_S_TMC_BLT_10_v1.3.0-115k.hex.zip)
[MEGA_S_v1.3.0-115k.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7217482/MEGA_S_v1.3.0-115k.hex.zip)

Best
Olli

Daddycool2708, 2021-09-23:
Hi Olli

I just tested the 115k version, but it is not working. I get the same error. I attach a screenshot from my octoprint.

[image]

knutwurst, 2021-09-23:
@Daddycool2708 : The Octoprint speed should be 250k as before. The 115k only applies to the Wifi Module, not usb!

Daddycool2708, 2021-09-23:
I did set it back to 250k but the result is the same. I get timeouts.

knutwurst, 2021-09-23:
last one.. I promise!
If this doesn't work with Octoprint, I'll disable Wifi per default.

[MEGA_S_TMC_BLT_10_v1.3.0-250k.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7218734/MEGA_S_TMC_BLT_10_v1.3.0-250k.hex.zip)
[MEGA_S_v1.3.0-250k.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7218735/MEGA_S_v1.3.0-250k.hex.zip)

Daddycool2708, 2021-09-23:
no not working from me, sorry 😢

knutwurst, 2021-09-23:
Fixed in Release [1.3.1](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.3.1)

closed 2021-09-23 by knutwurst

markecskace, 2021-09-23:
Hello! Sorry for my late response.
I went straight for MEGA_S_TMC_BLT_10_v1.3.0-250k.hex.zip and it is also not working for me.
Thank you for version 1.3.1!