knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #205: [BUG] Lesbarkeit Display + Manuelles bewegen der Achsen

wassereimer86, 2021-09-22:
Moin,

wenn ein Druck fertig ist, ist die verstrichene Zeit schlecht lesbar wenn man von der Seite schaut. Weiß verschwindet da und ist nur aus einem geraden Blickwinkel zu erkennen. Schwarz ist, wie der Text "Printing done, took:", deutlich besser lesbar.

Was mir noch aufgefallen ist: Wenn man über das Menü (Tools - Axis) die Achsen bewegt, ignoriert er die Endstops und macht weiter. Ohne Rücksicht auf Verluste. Ist das Absicht?

Viele Grüße
Jens

knutwurst, 2021-09-23:
Moin Jens,
eeeeeigentlich ist Github ja eine englische Pattform und alle Bugtickets sind auch englisch. Ich wollt's nur sagen ;) 

Zum Display: Darauf hab ich keinen Einfluss. Siehe: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/FAQ-(deutsch)#wieso-zeigt-das-display-keine-fehlermeldungen-an-und-wieso-sieht-es-noch-genau-so-aus-wie-vorher

> Was mir noch aufgefallen ist: Wenn man über das Menü (Tools - Axis) die Achsen bewegt, ignoriert er die Endstops und macht weiter. Ohne Rücksicht auf Verluste. Ist das Absicht?

Ja. Das liegt an der grundsätzlichen Steuerung wie Marlin 2.0 arbeitet. Die Endstops werden nur und ausschließlich beim G28 verwendet - sonst nicht. Erst nach einem G28 sind die Positionen der Achsen bekannt. Ursprünglich habe ich auch das Fahren der Achsen komplett verboten, solange kein G28 erfolgte, doch im Normalfall nutzt man die Funktion im Menü zum Bewegen auch nur dann, wenn man den Druckkopf an eine bestimmte Stelle bewegen will, oder um zu testen, ob die Motoren richtig herum laufen. Ebenso, wenn irgendein Kabel klemmt etc... und genau hier wäre es von Nachteil, wenn man erst homen müsste. Also habe ich für genau dieses Menü bzw das manuelle bewegen die Endstops ignoriert (da sie nicht bekannt sind).

lg
Olli

wassereimer86, 2021-09-23:
Kein Problem - in Zukunft in Englisch. 👍 War nur zu faul und einige handhaben das auf Github so und andere so. 😄

Danke für die Infos. 🙂

closed 2021-09-23 by wassereimer86