knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #206: Z2_DRIVER_TYPE  missing?

XoBoR, 2021-09-23:
Hi, first of all Dankeschön for this nice working firmware, saved me a lot of time!

Can it be that you are missing the definition of the driver type for the second Z driver, at least the MegaX has two Z axis steppers.

Esquilo007, 2021-09-23:
It should not be Z2, but E1 as far as I know for the second driver

XoBoR, 2021-09-23:
> It should not be Z2, but E1 as far as I know for the second driver

It's indeed connected to the Port labeled E1, but Marlin should still treat it as Z2 through the pin definition, I think.

knutwurst, 2021-09-25:
> It's indeed connected to the Port labeled E1, but Marlin should still treat it as Z2 through the pin definition, I think.

Nope, the pin definitions indicate which ports the board has. What this port is used for will be decided later. So you have to define for the E1 port what is installed there, if you use it. In the pin definitions, the Trigorilla Board does not have a Z2.

XoBoR, 2021-09-25:
> Nope, the pin definitions indicate which ports the board has. What this port is used for will be decided later. So you have to define for the E1 port what is installed there, if you use it. In the [...]

thanks for the clarification! 

I'm gonna close the issue.

closed 2021-09-25 by XoBoR