knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #207: [BUG] Falscher Thermistor

Oderik, 2021-09-25:
Ich möchte das Thema aus #20 noch Mal aufgreifen.

Ich habe mir einen neuen Thermistor gekauft und wollte herausfinden, welche Einstellung der erfordert. Deshalb habe ich viel herumgemessen und dabei festgestellt, dass die Messwerte des original Thermistors stark von den tatsächlichen Temperaturen abweichen.

Ich bin wie folgt vorgegangen: E3D V6 Heizblock (auf den bin ich gerade umgestiegen und habe "nur Probleme" 🙄) hat eine durchgehende Thermistoraufnahme. Daher habe ich auf der einen Seite den original Thermistor eingesteckt und auf der anderen Seite den Temperaturfühler meines Multimeters. Dann bin ich verschiedene Temperaturen angefahren und habe die Multimeter-Temperaturen notiert. Dasselbe Spiel habe ich auch mit der Kombination original Thermistor und mein neu gekaufter Thermistor (NTC 100k 3950) gemacht. Statt der Temperatur habe ich den Widerstand des Termistors beobachtet.

Die vom Multimeter gemessenen Temperaturen passen zu den entsprechend gemessenen Widerständen basierend auf diesem Datenblatt: https://hc-maschinentechnik.de/bilder/Datenblatt/Thermistor-Datenblatt.pdf Daher gehe ich davon aus, dass diese Messwerte belastbar sind.
Die vom Drucker angezeigten Temperaturen weichen jedoch deutlich ab:

Multimeter / °C | Original Thermistor / °C | Neuer Thermistor / Ω
---- | ---- | ----
70|60|17700
101|84|6900
121|100|3800
182|150|842
239|200|274
290|250|115

Vergleicht man die Temperaturen mit den Korrekturdaten aus Thermistor Presets 5 und 11 stellt man fest: Preset 11 ist tatsächlich das richtige.

Bedenkt man, dass Temperaturen ab 250°C problematisch für den Teflonschlauch sind, finde ich es etwas gruselig, bereits bei knapp unter 250°C gedruckt zu haben - das sind nämlich tatsächlich knapp 290°C.

Ich kann mir allerdings noch nicht erklären, wieso dieser deutliche Unterschied bei den bisherigen Messungen nicht aufgefallen sind. Gibt es womöglich unterschiedliche Varianten des Druckers bezüglich des mitgelieferten Thermistors?

@knutwurst, wenn ich es richtig verstanden habe, kann man den Thermistor per G-Code konfigurieren, wenn man Preset 1000 in der Firmware setzt. Kannst Du vielleicht die Firmware auf diesen Wert setzen und M305 P11 (oder meinetwegen P5) als Voreinstellung aufnehmen? Dann kann jeder selbst entscheiden, wie sein Thermistor interpretiert werden soll. Und ich kann meinen neuen Thermistor vernünftig benutzen. 😁

Oderik, 2021-09-26:
Das mit Preset 1000 war Unsinn - mit P wählt man den Sensor (P0) und nicht das Preset.

Es tut nichts zur Sache bzgl. der Einstellung des originalen Thermistors, aber für Nutzer dieses Thermistors sind vielleicht meine Einstellungen interessant: https://www.amazon.de/gp/product/B0951Y47HQ
Ich habe die Firmware mit `TEMP_SENSOR_0 1000` kompiliert. Per G-Code habe ich dann den Thermistor anhand der obigen Tabelle und dieser Seite eingestellt: https://configtool.reprapfirmware.org/Heaters
Der resultierende Befehl ist `M305 P0 T94500 B5442 C0.0000002470687` (danach abspeichern mit `M500`). Damit zeigt dann der Drucker praktisch dieselben Temperaturwerte wie mein Multimeter - insbesondere im Bereich über 200°C.

miscrtnone, 2021-10-02:
Es gibt zudem noch größere Abweichungen zwischen den einzelnen MEGAS selbst, mit gleichen Temperatur --> Widerstandzuordnung. Hier ein Beispiel von 2 verschiedenen  Mega S (gemessen in der Düse der jeweiligen Megas mittels Einstichthermomter der deutschen Firma Greisinger) mit gleicher Zuordnung, gemeinsim ist nur, dass sie nach oben abweichen.

Wenn man es genau haben möchte, dann muss man sich wahrscheinlich um die Temperatur für jeden einzelnen Mega individuell kümmern. Leider sind die Mega hier nicht so genau wie ich es mir wünschen würde.

Auch hier wird dies schon sehr negativ beschrieben für den Mega Pro (die Knutwurstfirmware kann also vermutlich nichts dafür, der Hersteller macht dies schon so: 

https://drucktipps3d.de/anycubic-mega-pro-neu-ab-mitte-august/

Aber jetzt zum Pragmatischen:
Würde man aber jetzt in die Firmware niedrigere Werte hinterlegen (zumindest um sich anzuhähern), dann kämen viele bei ihren derzeitigen Filamentsettings womöglich ins Strauchen, weil zu kalt und dann wird der Extruder ins Straucheln kommen.

Ich vermute, es ist daher besser per Default die höhere Temperatur zu lassen oder ganz explizit darauf hinzuweisen.

Anbei zwei exemplarische Messungen:
[image]
[image]

Oderik, 2021-10-03:
@miscrtnone danke für die Dreingabe!

Ich sehe es auch so, dass man die aktuellen Nutzer der Firmware letztlich nur vor Probleme stellt, wenn man jetzt plötzlich andere Thermistoreinstellungen verteilt. Und die bereits zahlreichen bestehenden Releases um eine Dimension mit alternativen Einstellungen zu erweitern (und damit praktisch zu verdoppeln) ist im Sinne der Übersichtlichkeit der Releases auch nicht förderlich.

Zumindest könnte @knutwurst aber auf die grundsätzliche Problematik aufmerksam machen, da die Firmware soweit ich das einschätzen kann recht verbreitet ist und daher eine gewisse Reichweite genießt.

Wie schon gesagt: letztlich ist ein korrekter Temperaturwert zweitrangig, wenn man ohnehin mit irgendwelchen Testdrucken eine geeignete Temperatur ermittelt. Nur das Thema PTFE-Schlauch bereitet mir etwas Sorge. Der sitzt im original Hotend dicht and der Düse und macht die Düsentemperatur dort mehr oder weniger mit. Da sollte man schon wissen, dass bei _tatsächlichen_ 250°C Schluss sein sollte, und dass man sich darauf ohne Kalibrierung nicht verlassen kann.

knutwurst, 2022-01-21:
Fixed in 7ef2dbd

sorry I haven't looked into it for so long :) 

Best,
Olli

closed 2022-01-21 by knutwurst