knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #208: [BUG] Wrong/uncomfortable Y-Position if aborting print

CompMartin, 2021-09-27:
### Bug Description

If aborting a direct print over USB in Cura, the Anycubic Mega S sends the ultrabase complete to the back. To remove the already printed stuff, this is not comfortable

### Steps to Reproduce

1. Print anything with Cura via USB
2. Abort the print in Cura
3. Look what is happening ;-)

Expected behavior: 
Move the base to the front the same way as for a succesful print

Actual behavior:
The base is moved to the back

v-a-d-e-r, 2021-09-27:
Haha, that's exact what I was thinking about too and did not understand. To remove all the stuff from the bed the x-axle is in the way....

HaGueRe, 2021-09-27:
I believe this is not a firmwarebug but a feature in Cura. If i abort a print with octopi, all axis are holding position and the heaters are going down.
Perhaps Cura wants to remind you to let cool down the Bed before removing the print.

lg Hans

CompMartin, 2021-09-27:
Hi Hans.
That wouln't make sense, as a correct finished part is moved to the front

Regards,
Martin

Jack77777777, 2021-09-29:
Did you open an issue at Cura ?

CompMartin, 2021-09-29:
I'm not that deep inside to decide wether it is a bug in Cura or in Marlin...
So you think it is inside Cura?

Jack77777777, 2021-09-29:
I think so. If i cancel a print on the lcd, octoprint or octodash the bed and nozzle stays at the point where i cancelled. I manually move the head upwards after cancel and all is fine 😁.

Jack77777777, 2021-09-29:
https://3dprinting.stackexchange.com/questions/8184/move-print-bed-to-front-at-the-end-of-a-print-rather-than-the-back/8185

CompMartin, 2021-09-29:
Yes, but this is different, as I understand.
If the print finishes correctly, the bed is moved to the front (ok, basically it is first moved back and then immediately to the front).
In case of aborting the print it is only moved to the back.

Jack77777777, 2021-09-29:
Start a print from SD Card, abort via terminal and see what happens.

https://marlinfw.org/docs/gcode/M524.html

CompMartin, 2021-09-29:
Just tested...then it is directly moved to the front...
So the complete behaviour is managed from Cura, I agree.

closed 2021-11-04 by github-actions[bot]