knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #209: Mega (S) underextrusion

mrrioes, 2021-10-02:
###Mein Setup
I3 Mega mit 2 Endstops
Trigorilla 1.0
Titan Klon nachgerüstet

### Bug Description
Zunächst habe ich die Firmware für den Mega S benutzt. Hier musste ich die E-Steps auf ca 950 einstellen, damit genug Material gefördert wird. Dann knarrt das Ding fürchterlich beim retracten und fährt dabei auch nicht.
Danach die Firmware für den I3 Mega classic benutzt, E-Steps auf ca. 400 eingestellt. Gleiches Verhalten.

Benutze sonst diese Firmware: https://www.thingiverse.com/thing:4358979
Dort passt alles bei ca. 400 Steps.

miscrtnone, 2021-10-02:
Da ich das Problem so nicht habe auf meinen Megas (Classic + Mega S + X) habe und die E-Steps so Pi mal Daumen passen, vermute ich eher, dass es weniger an der Firmware liegt, als viel mehr an anderen Faktoren, sprich (Düse zu kalt für das Filament und/oder Durchflussmenge, Verengung an anderer Stelle, ...). Gerade die mehr als Verdopplung der E-Steps für ein gewünschtes Ergebnis spricht dafür.

mrrioes, 2021-10-02:
Habe ich erst auch gedacht, hab aber heute morgen die alte Firmware wieder installiert und die Kiste hat danach direkt wieder wie erwartet funktioniert.
Mir fällt grad ein, dass ich vergessen hab zu erwähnen, dass ich tmc 2208 Treiber verwende (hab natürlich die passende Firmware benutzt), evtl hat es ja was damit zu tun. Wenn ich mich Recht entsinne, habe ich die damals so auf ca. 1,1V eingestellt.

miscrtnone, 2021-10-02:
Die TMC 2208 habe ich ebenso installiert allerdings sind Extruder bei mir die Standardversion der jeweiligen Megas (3 verschiedene also). Habe so keine Erklärung dafür bis. Einen erneuten Test auf Reproduzierbarkeit des Verhaltens, unter möglichst gleichen Umweltbedingungen (selbes Filament, Tag, Anpressdruck des Extruders) erscheint mir sinnvoll.

closed 2021-11-11 by github-actions[bot]