knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #210: [DOC] BLTouch wiring for Mega S/P

stklcode, 2021-10-03:
I recently added a _BLTouch_ sensor so my _Mega P_. The Mega P and - according to the PCB print - also the Mega S share a different hotend PCB which makes it even more elegant to add the sensor. There is an additional pinheader labelled "SWITCH" which is used for the piezo leveling helper. This is not suitable.

But the PCB features an "X-STOP" labelled slot for another pinheader. It's 2mm spacing, but with little bending a 2.54mm pinheader fits perfectly, if you want to change the original BLTouch cable.

[image]

This header is connected to A5 and A6/B6. The Pin we need is B5.

B5 is not connected on this PCB, so it's safe to swap A5 and B5 in the green connector. Doing so you can use the Knutwurst as is. Otherwise you would have to change the pin in the firmware configuration.

[image]

Works perfectly without any change to this FW.

[image]

poulsp, 2021-10-04:
Hi Stefan,

Good catch and info.
Can you tell me where and which pin two change in the firmware configuration?
I'm a fumble finger when it comes to changing the pins in the green connector. :)

regards Poul

stklcode, 2021-10-04:
You have to redefine the `Z_MIN_PROBE_PIN` pin.
This has been modified in the _pins_RAMPS.h_ in this project (don't know why, I'd just have redefined it in the configuraitn):
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/d84516016372b094183a3796ac8fb76fb182a18b/Marlin/src/pins/ramps/pins_RAMPS.h#L116

The current value is 2.

If I got it right, this should be swapped with the "FAN 1" Pin (7).

**I did not verify this!**
Apparently I didn't write it down anywhere, as I swapped the hardware pins before I ever had a look into the code. Feed free to try on own risk. I probably won't have an open printer to check it myself before the weekend

knutwurst, 2021-11-04:
Somehow I missed this ticket. Sorry The question is whether the PCB always looks like this on the Mega Pro, or whether Mega S PCBs are also installed.
If it always looks like yours, I would incorporate the pin change for the Mega Pro into the source code and adapt the documentation.

Can I use your picture for this?

stklcode, 2021-11-04:
I received two feedbacks of people with Mega P and the very same Hotend PCB.

Feel free to use the pictures.

A second change:
The Trigorilla clone seems to be different, too. My printer got "v0.0.2" which has servo-ports "S1" - "S4" exposed (pinheader only at S1) instead of S0-S3 in the wiki.
Other users report "v0.0.3" where the servo ports are different again. There's a 3x3 pinheader array soldered at the top center, where "5V GND S1" are the correct pins.

Both boards work well with the _MEGA_P_BLT_10_ target.

knutwurst, 2021-11-04:
The documentation is updated and I put your name in it if this is ok? Thank you very much!

Both other Trigorilla-Boards are already covered in the [Beginner's Guide](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)) or [Einsteiger Leitfaden](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(Deutsch)). 

Maybe I should reference the one guide in the other ;)

stklcode, 2021-11-04:
> The documentation is updated and I put your name in it if this is ok?

My name's already here, so again, feel free to use it.

> other Trigorilla-Boards are already covered in the [Beginner's Guide](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)) or [Einsteiger Leitfaden](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(Deutsch)).

Second link in the README and I obviously missed that one, nevermind :see_no_evil: Difference was probably too obvious.

closed 2021-11-05 by knutwurst

Kasu81, 2021-11-15:
Dear Knutwurst,

I own a I3 Mega Pro. With Board v.0.0.2.

am I right, if I will use H1 and H2 —> I have to change the wires from A5 to B5 and B5 to A5?

(zwei Lösungen dokumentiert um den BL Touch Sensor elegant am MEGA Pro anzuschließen. Hierbei kann der freie "X-Stop" Anschluss verwendet werden, wenn man entweder im oberen grünen Stecker die zwei Leitungen A5 und B5 wechselt)

weihan-ott, 2021-12-09:
> You have to redefine the `Z_MIN_PROBE_PIN` pin. This has been modified in the _pins_RAMPS.h_ in this project (don't know why, I'd have redefined it in the [configuration](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/d84516016372b094183a3796ac8fb76fb182a18b/Marlin/Configuration.h#L1128-L1132), but any way works fine):
> 
> https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/d84516016372b094183a3796ac8fb76fb182a18b/Marlin/src/pins/ramps/pins_RAMPS.h#L116
> 
> The current value is 2.
> 
> The one to swap with should be the `X_STOP_PIN` (3).
> 
> Don’t know why, but I initially had "FAN 1" Pin (7) in mind…
> 
> **I did not verify this!** Apparently I didn't write it down anywhere, as I swapped the hardware pins before I ever had a look into the code. Feed free to try on own risk. I probably won't have an open printer to check it myself before the weekend

I comfirm this change is wrong.MEGA PRO must swap A5 and B5

mnggh, 2022-01-04:
Hi @stklcode,
Thanks for your contribution as I’m about to install a BLTouch on my Mega Pro.

You state in your first post:
> There is an additional pinheader labelled "SWITCH" which is used for the piezo leveling helper. This is not suitable.

Can you please provide an explanation to this, why it’s not suitable?
I was wondering if you could also swap some cables or change the pin in the firmware config using the SWITCH pins.
Thanks.

stklcode, 2022-01-04:
See the pinout picture above. The “switch“ port uses shared pins with The T0 header. It simply provides power for the piezo buzzer that is shipped with the Mega P, so no usable signal is actually routed to the printer that can be switch via firmware.

Of course it is theoretically possible to use the pins. One might cut the lanes to A/B7 on the PCB and solder a wire to B5. Piezo leveling is obviously lost then.

I would rather cut both lanes and rewire it to the hortend pins to connect the laser module.

mnggh, 2022-01-05:
Ah, I see. Appreciate your explanation.

Your original solution is much easier and less harmful to the PCB. :)

Thanks!

mnggh, 2022-01-05:
Hi, me again.

I know what to do to make it work now. So far, so good.

But I’m lacking some comprehension. I understood, we need B5. And according this thread and RAMPS.h, B5 maps to pin 2 apparently.
Is there some kind of mapping table available or how do we know B5 maps to pin 2?

Sorry, if this is getting to off-topic, but I’m eager to also understand the backgrounds. :)

stklcode, 2022-01-05:
You can follow the wire and see/measure where it ends. For the 0.0.2 board (Trigorilla 1.4 clone) this is the “X+“ header of the endstops block in the southeast corner. There are a few pinout diagrams around the web where you will see it is pin D2. Otherwise you might look further and follow the connection from the pinheaded to the AtMega.

The i3 Mega has only one X endstop that is positioned on the frame, so the second pin is unused.

A5 and B5 are intended for alternative X limit switches on the hotend assembly. There are several hotend PCBs around, ours has connections on A5. This one is either not connected on the other end or shared with the actual limit switch, so it cannot trivially be swapped in the firmware.

mnggh, 2022-01-05:
Thanks!
Really interesting. Wow, I can imagine, following and / or measuring can be fuzzy. So I was thinking of some kind of official datasheet or something. :)

Yeah, I found a few PinPoint diagrams, but was confused with different boards and this different layouts. In addition, I wasn’t sure if „D2“ equals pin 2. Thanks for confirmation!

So B5 is actually an extension of D2 to the hotend PCB?
If so, I could also use D2 on the board directly and would not need to swap A5 / B5? Nor would I need to change anything in the firmware?

If yes to all, we could easily use one of these https://www.3djake.de/antclabs/bltouch-verlaengerungskabel-sm-du, swap only red and brown for the servo and that’s it. No soldering, no swapping cables on the PCB connector, no changing the firmware. Right? :)

stklcode, 2022-01-06:
That’s correct. You can directly connect 5 long wires (2x GND, +5V, Servo 1, D2) on the Trigorilla board. E.g. rd/ye/bn to the 3pin servo header (as usual) and bk/wh to the endstops block.

on the 0.0.2 the wiring would be:
| … | X+ | X- |
| --- | --- | ---- |
| … | D2 (wh) | D3 |
| … | GND (bk) | GND |
| … | +5V | +5V |

mnggh, 2022-01-06:
I agree to the wiring. Thanks!

After opening my Pro, I found that the endstop pins are already occupied by the SD-TFT-Hub-Switch PCB. I think, I understand the complexity behind this now. :)

Appreciate all your help and explanations!

Kilandor, 2022-01-14:
Appreciate this as well.

I have the Mega Pro. I was able to get this setup and working. Sadly yes you can't wire the BL touch onto the board as just like mnggh said, the PCB for it has a daughter board that covers and uses all the pins.

However simply taking the wires for A5 / B5 I used a little needle to release the wires from the harness and swapped them.
Then soldered a pin connector onto x-stop on the extruder end and was able to get it all working flawlessly.
I did swap the brown/red cables for the BL touch as well.

Just a note, if you plan to run the wire for the servo through the wire bundle, the 1m extension cable is not long enough. Thankfully my buddy had a 1m extra he didn't need on hand, so I just took it and modified it to be an extender with the wire connectors that came with it extra.

mnggh, 2022-01-14:
Hi Kilandor,
thanks for your hint, using extra long wires is indeed useful.
Following Knutwurst's guide ([German](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/BLTouch-Installation-(deutsch)) and [English](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/BLTouch-Installation-(english))), he explains to use two extensions (1x 100 cm, 1x 50cm) which sum to 150 cm in total. As an alternative, the "native" BLtouch extension posted above does the same trick.

For connecting the BLtouch to the mainboard, there seems to be way. I was inspired by this [video](https://www.youtube.com/watch?v=b8yWIWXus1U), in which the creator uses pin 21 on the board. Afterwards he reconfigured and compiled the firmware using the new pin 21 for Z_MIN_PROBE_PIN.
I did the same and it seems to work.

HTH.

whopperg, 2022-02-11:
Hey Everyone!
Could someone take some pictures from the bltouch autoleveling menu form the mega Pro? I cannot find this entry.
The only thing i found was the special menu which only shows "auto leveling". if i try to run this, it crashes into the print bed after the bltouch selftest.

I did run the probe configration as recommended in (https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/BLTouch-Installation-(deutsch)) and could also see that the probe is triggered (M114 in Pronterface)

Firmware: MEGA_S_DGUS_TMC_BLT_10_v1.4.1
Printer: Mega Pro
Trigorilla Board: 0.0.2
HotendPCB: Mega S extruder_V0.0.3

`M280 P0 S10 ; pushes the pin down
M119 ; Report Status of endstops

Reporting endstop status
x_min: open
y_min: open
z_min: open
z2_min: open
z_probe: open

(manually trigger probe with a piece of paper f.e.)

M119 ; Report Status of endstops

Reporting endstop status
x_min: open
y_min: open
z_min: open
z2_min: open
z_probe: TRIGGERED

M280 P0 S160 ; Release alarm`

so - as i assume - the wireing and firmware is correct.
What am i doing wrong?

derN3rd, 2022-03-30:
I'm currently having issues with installing a BLTouch (Clone from Geeetech) into my printer with Klipper (I know this is not the Klipper repo, but maybe someone has an idea)

- Anycubic Mega Pro with Trigorilla 0.0.2 Board and a Mega S 0.0.3 Extruder PIN Board
- I did solder on two PIN connectors at the X-STOP as described in the first post and did swap the two PINs in the green connector (as D3, as sKuhLight described, is already in use for the x endstop)
- Control PIN seems to be correct, as the BLTouch lights up and does the init sequence of down, up, down, up
  ```
  control_pin: PB5
  ```
 - The mount I used is the one from knutwurst: https://www.thingiverse.com/thing:5030218/files
 
 But when I try to query the sensor_pin to ^PE4 (which should be [Digital PIN 2](https://docs.arduino.cc/static/ed4d33b86bac66d54959fbb186795a11/54c3a/PinMap2560big_Rev2.png)) I only get 'open' state back.
 
 @sKuhLight did you need to change other things as well or configure something else?
 
 Thanks in advance