knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #221: [FR] Anycubic 4 max Pro version 1

kennethgomez01, 2021-10-31:
Hi, Just wondering if this firmware support for 4 max pro v1 not 2? Does this also has bltouch enabled already?

hanebuechenes, 2021-11-01:
I read that i can use it for 4max pro 1? How?

kennethgomez01, 2021-11-01:
> I read that i can use it for 4max pro 1? How?

Sorry what I mean, is this firmware works to any cubic 4 max pro?. Anyhow I have a firmware that works for 4 max pro v1.

hanebuechenes, 2021-11-01:
I think you can use this for all 4max versions, but maybe you need some modifications.. i dont know. i have V1 and want to have this firmware on it. do you have this 1.3.1 version here for 4max 1?

kennethgomez01, 2021-11-01:
@hanebuechenes I have the marlin 2.0. Let me give you a tip since this does not support the z offset wizzard for 4max pro(spent almost 2 weeks to do z offset) , don't trust videos(anycubic i3) they are not helpfl, you need to trust your eyesight.

hanebuechenes, 2021-11-01:
So i tried the version here 1.3.1, i found it as a compiled file. Seems to work good only the Y axis is wrong dierction, have to change this.

kennethgomez01, 2021-11-01:
the firmware I use has no problem. Take a note that the wiring must not replace the z limit switch.

kennethgomez01, 2021-11-02:
@hanebuechenes how did you solve the z-axis going down and stuck there when pressing "Home All" ?

hanebuechenes, 2021-11-02:
Compiled new. Change the value Form Y Axis. Load the Standard values after Install in Print Special Menü!

kennethgomez01, 2021-11-02:
@hanebuechenes If its ok can you send me the actual line code?

kennethgomez01, 2021-11-02:
because I don't know whic Y axis from the code are you refering too.

hanebuechenes, 2021-11-02:
So, i changed:
configuration.h
Line 1765 ff. for the printbedsize
Line 1533 for the Y axis direction

kennethgomez01, 2021-11-02:
@hanebuechenes to clarify this solved the z axis(up & down of the bed)?

kennethgomez01, 2021-11-02:
can you copy the actual change here please.

hanebuechenes, 2021-11-02:
I dont know exactly anymore.. Sorry. I think, the loading of the standard values and saving this to eeprom solved this problem... Maybe it is because of the wrong endstop is coded wrong..

hanebuechenes, 2021-11-02:
For changing Y direction (or other directions, if wrong):

#if ENABLED(KNUTWURST_4MAX)
        // Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
        #define INVERT_X_DIR false // set to true for stock drivers or TMC2208 with reversed connectors
        #define INVERT_Y_DIR **true** // set to false for stock drivers or TMC2208 with reversed connectors
        #define INVERT_Z_DIR false // set to false for stock drivers or TMC2208 with reversed connectors

hanebuechenes, 2021-11-02:
Maybe you must look here:

// @section homing

//#define NO_MOTION_BEFORE_HOMING // Inhibit movement until all axes have been homed

//#define UNKNOWN_Z_NO_RAISE      // Don't raise Z (lower the bed) if Z is "unknown." For beds that fall when Z is powered off.

//#define Z_HOMING_HEIGHT  4      // (mm) Minimal Z height before homing (G28) for Z clearance above the bed, clamps, ...
                                  // Be sure to have this much clearance over your Z_MAX_POS to prevent grinding.

//#define Z_AFTER_HOMING  10      // (mm) Height to move to after homing Z

// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

kennethgomez01, 2021-11-02:
so how does bltouch do to your print from 4 mac pro? Thanks for the info.

kennethgomez01, 2021-11-02:
@hanebuechenes  do you still have the copy of config file that works to you? Sorry its allot to ask, for this type of printer its hard to find for support

hanebuechenes, 2021-11-02:
Yes, i think my actual file is running fine!

kennethgomez01, 2021-11-02:
can you sent it through here if its ok @hanebuechenes ?

hanebuechenes, 2021-11-02:
[Configuration.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7458256/Configuration.zip)

kennethgomez01, 2021-11-02:
thank you very much!

hanebuechenes, 2021-11-02:
But remember to load factory defaults and save them to eeprom!

kennethgomez01, 2021-11-02:
I mean I have the marlin 2.0 working with bltouch is just that the z offset is hard to config

kennethgomez01, 2021-11-02:
I've been doing z-offset for 2 weeks, in some point in the bed the prints is not sticking.

kennethgomez01, 2021-11-02:
@hanebuechenes what is your wiring looks like?

hanebuechenes, 2021-11-02:
Ok.. i only made 4 point leveling and my prints are perfect!

hanebuechenes, 2021-11-02:
My printer is complete stock. No changes.

kennethgomez01, 2021-11-02:
@hanebuechenes wiring of bltouch sorry not clarifying that.

hanebuechenes, 2021-11-02:
I dont have. i make manual leveling :) just oldschool.

kennethgomez01, 2021-11-02:
so why return to manual if bltouch works for you?

hanebuechenes, 2021-11-02:
No sorry, that was misunderstanding. I never had BLtouch running.

kennethgomez01, 2021-11-02:
Oh.

knutwurst, 2021-11-02:
@hanebuechenes can you tell me exactly what type of printer you have? I thought, the 4MAX had a different ürint bed size... Please have a look at the original Anycubic sources: https://github.com/ANYCUBIC-3D/4MAX/blob/9b36eb13a2f908d09cf3a486035a81b88a7bd576/Configuration.h#L758

So as I understand... Y must be inverted and the build volume is the same as the 4MAX Pro 2.0?

Thanks in advance. I'll reflect your changes in the next version ;)

hanebuechenes, 2021-11-02:
I have 4Max pro 1
In this files the bedsize is wrong, so i had to change. also the axis direction. everything else seems to work (till now)..

knutwurst, 2021-11-02:
Ahhh Wait... The 4MAX Pro 1 isn't the same as the 4MAX, right?

hanebuechenes, 2021-11-02:
.. i think it is. Its just the first version. And then came V2.
Maybe this version also would work. i will check the differences.
This one i have: [https://www.anycubic.com/products/anycubic-4max-pro-3d-printer?variant=32416392773763](https://www.anycubic.com/products/anycubic-4max-pro-3d-printer?variant=32416392773763)

knutwurst, 2021-11-02:
I thought there were 3 printers of the 4MAX family... 
4MAX = Something weird
4MAX Pro v1 and v2 ... the v2 has a direct drive and a case light...

And that's the reason I did not manage to get all configurations right :D

hanebuechenes, 2021-11-02:
Ah ok :) so no, the V1 has the same case design. just some little changes, i dont know exactly now.

knutwurst, 2021-11-02:
This is the original 4MAX (non-pro!)
https://3duss.de/product/anycubic-4max-formax-3d-drucker/

[image]

hanebuechenes, 2021-11-02:
Oh Cool, this ine looks nice! Is it from metal?

knutwurst, 2021-11-02:
I think so. I don't know this printer :D 
But my firmware includes this configuration... 

4MAX (for the original non-pro)
4MAXP2 (for both, the Pro v1 and v2)

hanebuechenes, 2021-11-02:
Nice code you made, easy to compile, easy to understand. A lot of knowledge down in the comments! Thanks alot!

knutwurst, 2021-11-02:
Thanks very much!
But it's not just my merit. 3D printing lives from the community and without so many people like you who helped me with testing and figuring out configurations, it wouldn't have come this far :)

hanebuechenes, 2021-11-02:
Thats true, but your project is really cool. so if i can help i will. i have some i3 megas and this 4max. i like anycube printers because of the axis constructions (the old models). so lets see what will come.  i´m in this 3d printing stuff since 3 weeks, so :)

knutwurst, 2021-11-02:
Implemented in the current release 1.4.0 Beta 1. (See https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases)

kennethgomez01, 2021-11-02:
@knutwurst do you manage  to help me how to enable bltouch and configuration? Because somehow whenever I plug bltouch as z_switch, and press "Home" the bed is going down increments 2 and never move up, but probe is fown waiting to touch the surface.

kennethgomez01, 2021-11-02:
@knutwurst and complicated to communicate and understand this repository written  in Germannot English

Erik262, 2021-11-03:
Maybe I will contribute to the 4Max Pro V1 part incl. BLTouch Wiring documentation etc. Since knutwurst is already using the newest Marlin Firmware, so I don't have to do that part on my repository xD.

But I'm really wondering why the 4MPV1 (4 Max Pro Version 1) config is used as the same as the 4MPV2 config? 
And the settings, especially for the acceleration part, are pretty high, in my opinion. Or are these settings taken from alfrank?

@knutwurst Willst du mich vielleicht bitte in die Facebook-Gruppe beitreten lassen?

kennethgomez01, 2021-11-03:
@Erik262 what is the difference of this firmware and by pocketjogny firmware? Is these firmware more advance than johney version?

Erik262, 2021-11-04:
> @Erik262 what is the difference of this firmware and by pocketjogny firmware? Is these firmware more advance than johney version?

This here, has the newest stuff. Stick to this one. Have you already tried the given firmware here?

kennethgomez01, 2021-11-04:
@Erik262 no, I use pocketjohny, can you tell me which to enable config for 4max pro v1 including bltouch, as most are written in german.

kennethgomez01, 2021-11-05:
I can't compile its always telling me `                 from C:\Users\Kenneth Gomez\Downloads\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin\src\MarlinCore.cpp:31:
C:\Users\Kenneth Gomez\Downloads\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin\src\MarlinCore.cpp:306:52: error: size of array 'sensitive_pins' is not an integral constant-expression
     static const pin_t (&sensitive_pins)[pincount] PROGMEM = OnlyPins<SENSITIVE_PINS>::table;
                                                    ^
C:\Users\Kenneth Gomez\Downloads\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin\src\MarlinCore.cpp:306:85: error: could not convert template argument 'X_STEP_PIN' from '<type error>' to 'signed char'
     static const pin_t (&sensitive_pins)[pincount] PROGMEM = OnlyPins<SENSITIVE_PINS>::table;
                                                                                     ^
C:\Users\Kenneth Gomez\Downloads\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin\src\MarlinCore.cpp:306:85: error: could not convert template argument 'X_DIR_PIN' from '<type error>' to 'signed char'
C:\Users\Kenneth Gomez\Downloads\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin\src\MarlinCore.cpp:306:85: error: could not convert template argument 'Y_STEP_PIN' from '<type error>' to 'signed char'
C:\Users\Kenneth Gomez\Downloads\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin\src\MarlinCore.cpp:306:85: error: could not convert template argument 'Y_DIR_PIN' from '<type error>' to 'signed char'
C:\Users\Kenneth Gomez\Downloads\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin\src\MarlinCore.cpp:306:85: error: could not convert template argument 'Z_STEP_PIN' from '<type error>' to 'signed char'
C:\Users\Kenneth Gomez\Downloads\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin-2-0-x-Anycubic-i3-MEGA-S-master\Marlin\src\MarlinCore.cpp:306:85: error: could not convert template argument 'Z_DIR_PIN' from '<type error>' to 'signed char'
exit status 1
Error compiling for board Arduino Mega or Mega 2560.
`

Erik262, 2021-11-05:
@kennethgomez01 I could try to build a BLTouch Version with KnutWurst Firmware this weekend. **BUT I CANT** promise you if it works for you because I'm not near my printer until Christmas. So, therefore, I can't test it. At Christmas, I will try to build a BLTouch Version of the 4Max Pro V1. on my own machine and do a pull request on GitHub.

knutwurst, 2021-11-05:
@Erik262 it should be really straight forward when you know a suitable PIN for the Sensor (black/white wire). Everything else is already there within the KNUTWURST_BLTOUCH tree. I just didn't enable it, since I don't own a 4MAX Printer so I can not test it.

Erik262, 2021-11-05:
@knutwurst So you mean I only have to put the "-DKNUTWURST_BLTOUCH" flag from "build_unflags" to "build_flags" and I'm done and my printer is ready to touch the printbed? xD (of course if the offset is correct etc.)

I'm already in the group, but thanks!

knutwurst, 2021-11-05:
@Erik262 correct :D I mean... that "should" do the trick. It might need some more detailed configuration and at some points there are only default values for MEGA, MEGA_S etc but in general.. yes.

It also would be a good Idea to create a new PlatformIO profile like 4MAXP2_BLT.

Erik262, 2021-11-05:
@knutwurst If so, really nice work! 👌

knutwurst, 2021-11-05:
@Erik262 I just added the (completely untested!) BLTouch build environments. See 62479aa.
The fine tuning (offset, Pinout etc) is up to you :D

kennethgomez01, 2021-11-05:
@knutwurst thanks for reponse, I am trying every marlin version that works for my printer, since I am having very difficult weeks(2 weeks) straight calibarting my z offset, as some part of the the bed is not sticking at all.

[25 more replies]

closed 2021-12-16 by github-actions[bot]