knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #223: [BUG] 1.4.0-beta.1 wird von Cura 15.04.6 nicht mehr gefunden

v-a-d-e-r, 2021-11-02:
Im Gerätemanager steht der Drucker wie immer mit COM11. Geht aber nicht. Zurück auf 1.3.1 -> geht wieder.

smarti2019, 2021-11-02:
Gleiches Problem hier, nach Update und Load FW Default - Save Eprom keine Kommunikation mehr möglich.
Bei Pronterface kommt nur die Meldung das keine Kommunikation stattfinden kann. Falsche Baudrate.

Comport und Baudrate sind korrekt eingestellt.

knutwurst, 2021-11-03:
Ein bisschen mehr Informationen wären super :) 
Zum Beispiel, welcher Drucker das ist, welche Firmware genau geflasht wurde und ob ein Neustart etwas gebracht hat.

Danke :)

v-a-d-e-r, 2021-11-03:
Steht doch alles da, ein Any Mega S und 1.4.0-beta.1. ;-) Das alte Cura nur wegen Pronterface. Zum Glück konnte ich ich mit Cura 4.8.0 wieder zurück flashen auf die 1.3.1 unter Linux...

knutwurst, 2021-11-03:
[image]
Ah, als ich es das erste Mal gelesen hab (und als Mail bekommen) stand das da noch nicht :)

Und mit anderen Cura Versionen geht's? Nur die alte 15.04.6 geht nicht? Das ist sehr merkwürdig, aber ich schaue es mir mal an.

Pronterface gibt's übrigens als Standalone-Version.

smarti2019, 2021-11-03:
OK , Bei mir AnyCubic Mega-S TMC. Versuche mit Pronterface die Firmware zu flashen, Aktuell ist die 1.3.1 drauf. Und versuchte gestern die 1.4.0 Beta 1 zu flashen. Danach kommt kein Connect mehr zustande. Falsche Baudrate meldet Pronterface, egal was ich einstelle.

knutwurst, 2021-11-03:
[MEGA_S_TMC_140_Serial_Testfiles.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/7466507/MEGA_S_TMC_140_Serial_Testfiles.zip)
Bitte tut mir mal einen gefallen und testet diese beiden Firmwares für fen MEGA_S_TMC.

v-a-d-e-r, 2021-11-03:
Kann ich die einfach so nehmen ohne den Any zu schrotten? Ich benutze normalerweise die hier: MEGA_S_DGUS_TMC_BLT_10

knutwurst, 2021-11-03:
Und deswegen habe ich nach der genauen Version gefragt... 
Nehmen kannst du sie, aber der Sensor wird nicht funktionieren und das Special Menu auch nicht. Geht ja aber auch um USB. 

Defaults laden über Pronterface mit M501 und dann M500.

v-a-d-e-r, 2021-11-03:
Hab jetzt mal die MEGA_S_TMC_v1.4.0-b1_noserial drauf gemacht. Wird auch nicht erkannt. Aber wie flashe ich jetzt die andere Version, da keine Kommunikation mehr mit dem Drucker möglich ist!?

knutwurst, 2021-11-03:
Mit einer anderen Cura Version? Oder Prusaslicer oder XLoader? 
Die Aussage war doch, dass es nur mit Cura 15.4 nicht geht.

v-a-d-e-r, 2021-11-03:
Jetzt wird er nicht mal mehr unter Linux gefunden mit Cura 4.8.0

knutwurst, 2021-11-03:
Wurde denn erfolgreich geflasht?
Bitte versuch es mal mit 115200 baud. Notfalls unter Windows mit XLoader.

Wird im Gerätemanager noch ein Device erkannt?

v-a-d-e-r, 2021-11-03:
Im Gerätemanager ist er drin. Es kommt auch das "Beep" vom Drucker, wenn eine Verbindung hergestellt wird. Probiere jetzt gerade XLoader mit 115200....

v-a-d-e-r, 2021-11-03:
Ok, beide Files gehen nicht. Log:
Changing monitoring state from 'Detecting baudrate' to 'Closed'
Changing monitoring state from 'Closed' to 'Error: No more baudrates to test, and no s...'
Connection closed, closing down monitor

stklcode, 2021-11-03:
Spannend, ich habe ich ähnliche Probleme auf meinem Mega P mit dem 1.4.0-beta.1 Image.

Octoprint (1.6, im OpenShift Container) schreibt nur Zeichensalat in der Konsole raus, recht egal mit welcher Baudrate. Cura (4.11, Linux) braucht eine Sekunde länger als üblich zur Verbindung, funktioniert danach aber tadellos.

Trat nach dem Flash der 1.4 und dem Laden der FW Defaults auf (vorher war irgendwas experimentelles auf Stock Marlin Basis drauf). Immerhin ein reproduzierbares Problem.

SteFiD93, 2021-11-03:
I have the exact same problem with CHIRON_v1.4.0-b1.hex I tried below with 1.3.1 and it works perfectly, I can flash the firmware, but as soon as I put 1.4 it gives me connection problems, I know it's very strange but it happens

knutwurst, 2021-11-03:
It is strange actually. Does it occur directly when loading the defaults, or when saving afterwards?

Unfortunately, some custom code relating to the EEPROM was also lost during the changeover. I have a feeling that there is a connection.

However, much of this only relates to the Chiron. That's why it's strange. When I have the time, I'll try to debug what's going on.

stklcode, 2021-11-03:
Appears immediately after flashing without touching the EEPROM.

Flash 1.3.1, load/save EEPROM defaults, flash 1.4.0-b1 => communication broken.

smarti2019, 2021-11-03:
@knutwurst , ich teste den ober verlinkten Flash mal heute Abend nach Feierabend und gebe bescheid, ob es dann auf mein MEGA-S TMC funktioniert.

knutwurst, 2021-11-03:
@stklcode 
Unfortunately, I don't have a printer at hand at the moment and actually have to work in my "real job" ;) But I've committed in a few small changes. Can't imagine that has anything to do with it, but maybe you can test it?

I can't really imagine that it's really because of the serial communication. Does the printer behave differently? Is it slower than before?

@smarti2019 
Vermutlich ändert sich nicht viel. Das Problem scheint etwas tiefer zu gehen. Trotzdem danke!

stklcode, 2021-11-03:
Any reson why `MYSERIAL1.begin(BAUDRATE);` was removed?

No idea why i've marked this line as patch, but I guess without initialization of the first interface, it's definitely worse, i.e. there is no connection at all after flashing a FW with this patch.

knutwurst, 2021-11-03:
Waaaaait.... I'm missing MYSERIAL0 :D I thought that 1 is the second connection and since it was initialized with BAUDRATE_2 (which is set to 500k for Wifi communication).

Uhm.. Good point! I just wanted to know if you're paying attention :D

[image]

I just reverted and added the correct baudrate. This could be the issue in ne first place!
See: d581c26

stklcode, 2021-11-03:
I guess `MYSERIAL1`  is fine. Initialization changed between Marlin 2.0.8 and 2.0.9

```c
 MYSERIAL1.begin(BAUDRATE);
  millis_t serial_connect_timeout = millis() + 1000UL;
  while (!MYSERIAL1.connected() && PENDING(millis(), serial_connect_timeout)) { /*nada*/ }

  #if HAS_MULTI_SERIAL && !HAS_ETHERNET
    MYSERIAL2.begin(BAUDRATE);
/* [...] */
```

```c
  MYSERIAL1.begin(BAUDRATE);
  millis_t serial_connect_timeout = millis() + 1000UL;
  while (!MYSERIAL1.connected() && PENDING(millis(), serial_connect_timeout)) { /*nada*/ }

  #if HAS_MULTI_SERIAL && !HAS_ETHERNET
    #ifndef BAUDRATE_2
      #define BAUDRATE_2 BAUDRATE
    #endif
    MYSERIAL2.begin(BAUDRATE_2);
/* [...] */
```

knutwurst, 2021-11-03:
Yes, MYSERIAL1 is correct. In my previous implementation I added the second configurable interface for WIFI. Now in 2.0.9 there is a BAUDRATE_2 define already. So I got a little confused with your patch and removed the line.

So in 1.4.0 the default baudrate is 500k and afterwards there was no serial port at all. 

It should work now.

stklcode, 2021-11-03:
Out of curiosity, is flashing the Hex file via direct ISP connection safe :see_no_evil: ?

I should read code before flashing...

knutwurst, 2021-11-03:
Um, only if you don't flash at address 0x00 or include a bootloader in your binary.
The best way is actually to flash the bootloader via ISP. You can then install the firmware via USB.
You can get the right Atmega 2560 Arduino bootloader directly from the Arduino IDE, if you don't have one anymore.
If you flash the firmware without BL, you can no longer install a FW via USB after the firmware update.

stklcode, 2021-11-03:
OK, confirmed 1.4.0-beta.2 working on my Mega P. Both Cura and Octoprint communicate with the printer at 250k baud :smile: 

Re-Flashed via ISP (didn't use Arduino stuff for years, so just re-flashed with avrdude CLI).

smarti2019, 2021-11-03:
OK, just compiled the current version 1.4.0 Beta 2 and everything works perfectly.

great!

closed 2021-11-03 by knutwurst

kellogg76, 2021-11-21:
Can someone provide some guidance on how to flash 1.4.0-beta.3 over 1.4.0-beta.1 on a Mega X given that Cura cannot connect.
Xloader gives me two device options CC01 and CR03, I've tried both and the tool just sits at "Uploading...." and is unresponsive.