knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #224: 4 Max Pro V1 Thermisistor issue

kennethgomez01, 2021-11-07:
I was having a problem with thermisitor value fluctuate, even I change my thermisistor multiple times.

knutwurst, 2021-11-16:
Do you know what kind of thermistor the 4MAX Pro V1 uses? If it is different from the 4MAX Pro V2 there might be the issue :/ 

This is why it is still "Beta" ;) 

You could change the Thermistor in the Firmware or tra the 4MAX (non-pro) Firmware to test the temperature. Please be aware that your printer will not work correctly with the non-pro FW.

kennethgomez01, 2021-11-16:
@knutwurst I fixed this issue, mostly the error occur when sdcard is plug to the 3D printer, and mange it to work with octoprint alone. But thanks for response.

closed 2021-12-22 by github-actions[bot]