knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #226: [FR] (Does this firmware support 4max pro v2.0 BLT?)

jpmini, 2021-11-14:
Hello. Does this firmware support 4max pro v2.0 BLT?

I uploaded "4MAX P2 DGUS TMC BLT 10" on PlatformIO, but it doesn't work.
Z-axis leveling is not possible.

https://youtube.com/shorts/mb7xfhr1Mmo?feature=share

My 4MAX P2
-Anycubic 0.0.2 display
-Trigorilla 0.0.2
-All TMC2208
-BL-Touch
Will be.

Any advice would be appreciated.

kennethgomez01, 2021-11-16:
can you sent me picture from the board and from the extruder please. because mine is v1 which I believe has the same config.

jpmini, 2021-11-16:
Thank you for your reply.

I have installed BLTouch and saved it with an offset of X-27, Y-12, Z-0.3.
Replace BLT (white + black) with the original Z end stop.
Is the connection correct?

Thank you for your support.

> 
> 
>> 2021/11/16 14:33、kennethgomez01 ***@***.***>のメール:
>> 
>> can you sent me picture from the board and from the extruder please. because mine is v1 which I believe has the same config.
>>

kennethgomez01, 2021-11-16:
oh, ok is it ok can you sent me a picture of your wiring in the motherboard underneath the 3D printer.

jpmini, 2021-11-16:
Thank you for your reply.

Please tell me if you need a different angle photo!

Thanks for your support!

> 2021/11/16 16:33、kennethgomez01 ***@***.***>のメール:
> 
> oh, ok is it ok can you sent me a picture of your wiring in the motherboard underneath the 3D printer.

kennethgomez01, 2021-11-16:
capture all the angles, the better I can see the problem.

jpmini, 2021-11-16:
Thank you for your reply.
I sent you a photo.
Please let me know if you need another angle!

kennethgomez01, 2021-11-16:
@jpmini Sorry, can't see the attachment. Wouldn't you mind send it through GitHub not by email?

jpmini, 2021-11-16:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

jpmini, 2021-11-16:
Thank you for your reply.
I'm sorry I couldn't send it.
I've attached!

kennethgomez01, 2021-11-16:
try this firmware, can you show me the your config_adv and config? I believe the reason that bltouch is not working as z-switch is you did not uncomment #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN . I did have this problem before.

kennethgomez01, 2021-11-16:
@jpmini Do you have problem compiling?

jpmini, 2021-11-16:
Thank you for your reply.

I built and uploaded 4MAXP2 DGUS TMC BLT 10 of PROJECT TASKS of Marlin-2-0-x-Anycubic-i3-MEGA-S (1.4.0-beta.3), but how is config_adv and config? Can you see it
[image]
[image]
?

knutwurst, 2021-11-16:
@jpmini and @kennethgomez01 
I just updated the configuration and if you checkout (git clone) the current master branch, the BLTouch feature should work.

jpmini, 2021-11-16:
Thank you for your reply.
Thanks for your support!

How do I check out (git clone) the master branch?
I tried to find it, but I didn't understand ...

knutwurst, 2021-11-16:
It's just downloading the current source code. Normally you would download the sources from the assets in the download-section (where you can get the hex files). But now instead, use this
[image]
:

knutwurst, 2021-11-16:
Looks good!
If you don't mind, could you tell me the correct X, Y and Z offset afterwards so I can adapt the settings as default for the upcoming firmware versions? 

Thanks!

jpmini, 2021-11-16:
With my BL Touch mounting method, the offset is X-27 Y-12 Z-0.1.

I didn't want to make a mount, so I screwed the BLTouch directly.

[image]

It stops with a single screw, but it seems to be strong enough because it fits in the original mount.
If you're worried, put a plate on top of the BL Touch to make it even stronger.

[image]

[image]
[image]

[image]

I will test it further!
Thank you for your support! !! !!

jpmini, 2021-11-16:
There was one problem when testing.

When I run G29, the head interferes with the printer.

https://www.youtube.com/watch?v=XU1C6dEJjOA

Can this be fixed from the G-code?

kennethgomez01, 2021-11-16:
@jpmini no worry, I will try to make a video later, to show you how mine works.

knutwurst, 2021-11-17:
@jpmini you have 2 options:
1. Set the X offset a bit to the positive, so the printer "thinks" that the probe is a few mm more to the right, so it does not move so far to the right.

2. Increase the mesh inset (currently at 1mm. See https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/2719ba669071d7b62b49deebfc8f05a41690751f/Marlin/Configuration.h#L2334), but this has to be done in the firmware.

kennethgomez01, 2021-11-17:
@knutwurst ghanks for support, mine works fine after 3 weeks of tunning z_offset.[Result](https://storage.googleapis.com/prod-tsd-timelapses/private/27298717_tagged.mp4?Expires=1896344061&GoogleAccessId=appserviceaccount%40tsdtechnology.iam.gserviceaccount.com&Signature=vHTICnYIcsFOM2eQy3UGHHYdsmb%2BWNRNnGVWWraYrX%2BjVKnTXg3J0ugPaqhzWlyrgUCjTlZyUJyV1d9DnzUNB8Lz2DaNZ9IQoHG8Pv%2BdTVIClR56rli1sB5vpXekHRU50geSL1QdT7GnzDcJoZUw6twwceqc3ZsNY48%2Fk0rAbUP38oxjkHvVk2tY68Qa7A%2Fo%2FSuRfUgN93DS3eVXjZ1RPYrUmdzkSwQWIaI6cObPEO9mdvT5WQ5DZ98ef%2B%2FMFzCb3bfqjlGpQyy6CHKrCVO0SOZHekvvi6pdzYkkbr%2FhoZcs4AKvgo2C2c2n8brX4183UCcFTJe2ZI3n4gE62Km3nw%3D%3D)

jpmini, 2021-11-17:
Thank you for your reply.

I tried the X offset, but I changed X from -27 to -17, but it hits a little.

I am building and uploading 4MAXP DGUS TMC BLT 10 from PROJECT TASKS. How should I apply Configuration.h?

Thank you for your support
[image]
[image]

.

kennethgomez01, 2021-11-17:
Iz your bltouch installed to the left or right of the extruder? Also try to
make the grid-y and x point count 3. Remember that positive x is to the
right and -x to the left.

jpmini, 2021-11-18:
My bltouch is attached to the left and is offset to X-27 Y-12 Z-0.1.

kennethgomez01, 2021-11-18:
it would be better if you could take video of the actual problem that you are having, that way it was easy to know what wrong.

jpmini, 2021-11-18:
Thank you for your reply

https://www.youtube.com/watch?v=XU1C6dEJjOA

The video is here, but the extruder hits the printer. (0: 06 ~)

I changed #define MESH_INSET 10 to #define MESH_INSET 50, but it doesn't seem to change.

knutwurst, 2021-11-18:
I'm so sorry for leading you in the wrong direction :/ You are using Bilinear leveling, so your variable is called PROBING_MARGIN to narrow the grid.

kennethgomez01, 2021-11-18:
try to decrease 2 increment for x-offset for example 2.20 - 2 = new x offset is 2.18. Try that until the extruder and bltouch probe only goes to the bed As to what I can see its looks like a x offset problem.

kennethgomez01, 2021-11-18:
please each time you upload new firmware please do 'M500' and 'M5001' 'M5003'

jpmini, 2021-11-18:
Thank you for your support! !! !!

I tried setting the BLT offset to X-17 and Y-12, but it came off the bed.

I was able to avoid changing PROBING_MARGIN from 10 to 35 with a margin!

https://www.youtube.com/watch?v=L9-3Ys_1V7U

kennethgomez01, 2021-11-18:
@jpmini Good for you! Now its time to test out the z-offset, mine took 3 weeks of z offset calibration, until got succeed. [https://storage.googleapis.com/prod-tsd-timelapses/private/27298717_tagged.mp4?Expires=1896344061&GoogleAccessId=appserviceaccount%40tsdtechnology.iam.gserviceaccount.com&Signature=vHTICnYIcsFOM2eQy3UGHHYdsmb%2BWNRNnGVWWraYrX%2BjVKnTXg3J0ugPaqhzWlyrgUCjTlZyUJyV1d9DnzUNB8Lz2DaNZ9IQoHG8Pv%2BdTVIClR56rli1sB5vpXekHRU50geSL1QdT7GnzDcJoZUw6twwceqc3ZsNY48%2Fk0rAbUP38oxjkHvVk2tY68Qa7A%2Fo%2FSuRfUgN93DS3eVXjZ1RPYrUmdzkSwQWIaI6cObPEO9mdvT5WQ5DZ98ef%2B%2FMFzCb3bfqjlGpQyy6CHKrCVO0SOZHekvvi6pdzYkkbr%2FhoZcs4AKvgo2C2c2n8brX4183UCcFTJe2ZI3n4gE62Km3nw%3D%3D](url)

jpmini, 2021-11-18:
Thanks to the support, I got my first test print! !! !
[image]
!

kennethgomez01, 2021-11-20:
@jpmini how did you get the value for z-offset? What did you use to calibrate it?

oshimo, 2021-11-20:
Sorry to jump in, but I have followed along with this to set up my BLTouch. 
When running the 4 point, Z does not attempt to move up to the BLTouch. Am I missing a step here?

https://youtu.be/XR2e26T53Bw

kennethgomez01, 2021-11-20:
Yes, please be sure to uncomment #Z_min Probe Pin.

jpmini, 2021-11-20:
@oshimo The Z motor is flipping.

jpmini, 2021-11-20:
@kennethgomez01 I confirmed the offset value with M114.
Now that the nozzle has been replaced, the Z offset has changed slightly.

kennethgomez01, 2021-11-21:
@jpmini I mean how did you get it like, did you use paper and adjust z-offset manually?

oshimo, 2021-11-21:
Unfortunately it is the same result when uncommenting

kennethgomez01, 2021-11-21:
@oshimo are you using v1 or v2 Anycubic 4 Max pro?

oshimo, 2021-11-21:
V2

kennethgomez01, 2021-11-21:
When your compiling must choose 4maxp2_blt

oshimo, 2021-11-21:
Yes, I am using 4MAXP2_DGUS_TMC_BLT_10
I have confirmed I have no logo board.

kennethgomez01, 2021-11-21:
Dont use DGus its not going to work, try 4max_p2 blt

kennethgomez01, 2021-11-21:
4max_p2 blt_11 is the version your looking for your trigorilla 0.02(no logo
board)

AKharytonchyk, 2021-12-08:
Hi all, I am wondering whether there is an option to connect BLTouch via level port on extruder?

[image]

jklaus, 2022-01-04:
@AKharytonchyk - I am interested to know as well.  Please let me know if you figure it out.  My 4Max Pro V2.0 is being delivered today and I want to get bed leveling going ASAP.

AKharytonchyk, 2022-01-19:
@jklaus it looks like you will still need to pass one wire back to the controller board. This is because BLTouch needs a D11, aka servo control, aka signal, cable to connect to the board. This cable will control the BLTouch servo that will move the pin. 
[image]

bigmacreejr, 2022-01-19:
> With my BL Touch mounting method, the offset is X-27 Y-12 Z-0.1.
>
> I didn't want to make a mount, so I screwed the BLTouch directly.
>
> [image]
>
> It stops with a single screw, but it seems to be strong [...]

okay, im semi-new to 3d printing and have the same printer as you and an authentic bltouch and installed it the same way you have in your example pictures,  and installed it in my printer (it lights up and moves on boot) but for the life of me i can not get the printer to home correctly, when i go to the special menu and press start auto leveling, it acts normal until the xtruder and bltouch move the the left to home before it should start leveling, when it tried to go to the left and home it makes this awful noise and doesnt move, like its trying to move further left than it can. i can not seem to get it worked out, so if you can either share your firmware (since im using the exact same setup as you) or assist me in getting my firmware working then i would very much appreciate it

MasterYod4, 2022-02-09:
Hello you have to loosen the two screws on the left of the limit switch and move the limit switch a few mm to the right and screw it tight again.
I had the same problem and was able to solve it like this

closed 2022-03-16 by github-actions[bot]

JonnyHaystack, 2022-05-07:
> @oshimo 4max_p2 blt_11 is the version your looking for your trigorilla 0.02(no logo board)

I don't think this is correct..? At least if you're wiring your BLT to S0. BLT_11 is for the TriGorilla 1.1 which is the only one with a different servo pinout. The other boards should all be the same.

The servo port numbering is a bit confusing though. Knutwurst's wiki says this:
[image]
But it seems like the servo numbering (SERVO0, SERVO1, etc) matches neither the silkscreen nor the physical ordering of them, based on this pinout diagram https://cdn.thingiverse.com/assets/32/8c/6d/ab/5b/TriGoRiLLa_PINs.pdf

That diagram also doesn't show D7 being used for servos at all though, which makes me even more confused...

But from what I've read in here, D11 is the pin that we want to use if we wire to S0. If we used BLT_11 it would surely configure it to use D5 instead, which wouldn't work. Please correct me if I'm misunderstanding something. I'm currently setting up my BLTouch in Klipper, and will report back afterwards.

P.s. Klipper is working great on this printer so far! I'm going to try to replace the touchscreen with a 3.5" Raspberry Pi touchscreen using Klipperscreen as well, so that I can have an actual good touchscreen. Hopefully that works out too 😄