knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #227: [FR] Anycubic Chiron, Anycubic 1.0 Display, Bltouch

WesleyvH, 2021-11-17:
### Description

I need a firmware version for the Anycubic Chiron with the Anycubic 1.0 display and Bltouch enabled. I tried to make the config myself with PlatformIO but without succes.

### Feature Workflow

Getting Build errors

1. In i3_mega.ini added -DKNUTWURST_BLTOUCH to CHIRON build_flags and deleted it from build_unflags:

[env:CHIRON]
platform          = atmelavr
board             = megaatmega2560
board_build.f_cpu = 16000000L
lib_deps          = ${common.lib_deps}
  TMC26XStepper=https://github.com/trinamic/TMC26XStepper/archive/master.zip
src_filter        = ${common.default_src_filter} +<src/HAL/AVR>
build_flags       = ${common.build_flags} -DMOTHERBOARD=BOARD_TRIGORILLA_CHIRON -DKNUTWURST_CHIRON -DKNUTWURST_BLTOUCH
build_unflags     = -DKNUTWURST_MEGA -DKNUTWURST_MEGA_S -DKNUTWURST_MEGA_X -DKNUTWURST_MEGA_P -DKNUTWURST_DGUS2_TFT -DKNUTWURST_TMC -DKNUTWURST_ONE_Z_ENDSTOP -DKNUTWURST_4MAXP2 -DKNUTWURST_4MAX

2. Changed Nozzle to probe offset in configuration.h:

#if ENABLED(KNUTWURST_BLTOUCH)
  #define NOZZLE_TO_PROBE_OFFSET { -25, -3, 0 } //https://www.thingiverse.com/thing:2824005
  //#define NOZZLE_TO_PROBE_OFFSET { 29, -15, 0 } //X-Carriage
#endif

3. Clicked on project tasks -> CHIRON -> Build

4. Getting the following errors: 

Marlin\src\lcd\anycubic_touchscreen.cpp: In member function 'void AnycubicTouchscreenClass::PrintList()':
Marlin\src\lcd\anycubic_touchscreen.cpp:1072:9: error: duplicate case value
         case 4: // Page 2 for Chiron ABL
         ^~~~
Marlin\src\lcd\anycubic_touchscreen.cpp:1059:9: note: previously used here
         case 4: // Page 2 for BLTouch
         ^~~~
*** [.pio\build\CHIRON\src\src\lcd\anycubic_touchscreen.cpp.o] Error 1

#### Additional Information

It seems like i have to disable the Chiron_ABL but i can't seem to find out how.

knutwurst, 2021-11-17:
When you try to use the BLTouch, you can not use the internal ABL Feature. Look at the configuration.h and disable #define KNUTWURST_TFT_LEVELING which disables all internal ABL functionality.

WesleyvH, 2021-11-17:
Okay thanks with the #define KNUTWURST_TFT_LEVELING disabled it will compile, the bltouch is blinking red though and not working yet, this could be the pins configuration of the trigorilla board. Already switched the black and white cable but no luck with that.

WesleyvH, 2021-11-17:
Changed the servo pins in pins_TRIGORILLA_CHIRON.h to:

#if MB(TRIGORILLA_14_11)
  #define SERVO0_PIN                           11
  #define SERVO1_PIN                           6
  #define SERVO2_PIN                           5
  #define SERVO3_PIN                           4
#endif

Figured out the sensor was DOA took another one from the stash and works like a charm now.

WesleyvH, 2021-11-17:
Thank you very much for the help and the awesome firmware!

closed 2021-11-17 by WesleyvH