knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #228: [BUG] Resonant frequency hit on Z axis for probing on i3 mega X

JernejL, 2021-11-22:
### Bug Description

The currently configuired I3 mega X bltouch probing speed is set to an resonant frequency of the printer frame. 

### Steps to Reproduce

1. Run bltouch probing
2. Listen to noise

**Expected behavior:** Silent probing

**Actual behavior:** Everything is vibrating

#### Additional Information

F speed for Z axis, is if within 260 up to 360, printer will shake. 
I would recommend Z_PROBE_SPEED_SLOW to be 250 and Z_PROBE_SPEED_FAST 400 
 
I had this happening since the bltouch upgrade (Everything except steppers ( TMC2209 ) and bltouch was stock). 
Then i replaced the Z axis leadscrews twice (first BRASS, then i tried delrin / pom) and still had this occur, so i assumed that this could be something that was not taken into account when building the firmware. 

I don't know if this happens with everybody's printer but i will happily provide a video if this is not taken seriously.

JernejL, 2021-11-23:
Adding a video: 
 
https://youtu.be/lbwwsECMSPg 
 
Pay attention to the weird buzzing and vibrations when the Z axis goes up & down.

knutwurst, 2021-11-23:
Hi @JernejL , thank you very much! I don't own a Mega X so it's good to see one in action ;)
Yes... it seems that the Z feedrate is a bit too high. The probing speed is fine, but I would suggest lowering the Z feedrate with M203 from 18 (default) to 12 or something. If it works for you, let me know and I'll change the default value in the upcoming version.

Best,
Olli

JernejL, 2021-11-23:
I'm not sure if M203 even affects probing speeds, since it looks like this is somehow specific / hardcoded in marlin. 
 
In configuration.h there are constants for XY_PROBE_SPEED, Z_PROBE_SPEED_SLOW and Z_PROBE_SPEED_FAST 
 
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/f631eeac240aa47bcd440c62070389435d86e98c/Marlin/Configuration.h#L1561 
 
Seems like this was changed ( i guess in very latest syncup to upstream marlin) to XY_PROBE_FEEDRATE, Z_PROBE_FEEDRATE_FAST  and Z_PROBE_FEEDRATE_SLOW

closed 2021-12-29 by github-actions[bot]