knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #229: [BUG] Head Nozzle crash during probing (Serious) (Video)

JernejL, 2021-11-23:
### Bug Description

I've followed the FAQ and succesfully upgraded i3 mega X to bltouch in june ( https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/149 ) . I also helped others with their wiring so i know that to be good. 
 
On the printer i use MEGA_X_DGUS_BLT_10 but it is admittedly a few releases older version, i was planning to upgrade this week. 
 
When making a video to show the oscillation / vibration during probing i also captured a head crash which has not happened ever to me before. 
 
It seems like the head probed a point correctly, then went into error / red light state and then marlin continued with the next probe without the probe extended, going even below the Z endstop, crashing into the bed, severely damaging it in the center. 
 
https://www.youtube.com/watch?v=ErtHUwqIawc
 
I am accepting that this is by big chance a failed bltouch wiring (bltouch endstop black & white wires probably), and i will inspect and report, but that this even happened means that at least 2 safeties failed: 
 
- **The probing moved head below the homed Z limit. The i3 mega X has two Z axis endstops and during probing it didn't seem to check or obey this?** 
- **BLtouch error state was not taken into account and cancel probing, it continued to next point (and then crashed head into bed)**
 
### Steps to Reproduce

1. Start probing
2. Probe indicates error
3. Probe crashes regardless. 

**Expected behavior:** Probing would detect error and cease with an error state. 

**Actual behavior:** Head crashes. 

#### Additional Information

Video indicating the crash: 
 
https://www.youtube.com/watch?v=ErtHUwqIawc

knutwurst, 2021-11-23:
OMG! That's not good :/ 
I'm so sorry!

You wrote
> On the printer i use MEGA_X_DGUS_BLT_10
What version are you using?

JernejL, 2021-11-23:
I think i used MEGA_X_DGUS_BLT_10_v1.2.0.hex from 1.2.0 since i downloaded it on 17. July. 
 
It should be this: 
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.2.0
 
I know i'm quite a few versions behind, i was planning to compile my own firmware and do stuff like increase GRID_MAX_POINTS_X from 5 to 7 because i have a part of hotbed between probe points and it seems to be always missed during compensation.

kennethgomez01, 2021-11-23:
@JernejL don't use DGUS , its not going to work. You should use XXX_blt.x(10 for standard trigorilla board while 0.02 no logo for 11).hex

JernejL, 2021-11-23:
@kennethgomez01 I have a i3 mega **X** and I need to use DGUS as that is the version for my printer LCD, i'm using printer standalone without a computer or octoprint attached to it. I used this without an incident since July. 
 
My printer uses the blue/yellow LCD and that is the only version that makes special menu show up. 
 
See: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Types-of-Anycubic-Touchscreens

kennethgomez01, 2021-11-23:
@JernejL my bad sorry, maybe wiring or not enable z_min_pin?

JernejL, 2021-11-23:
@kennethgomez01 : It worked fine for 5 months, i'm guessing wiring. 
 
The reason for ticket is that i'm sure there are several ways this could been prevented, i'd like to understand why this happened, why the Z endstops and error state from bltouch were not handled.

kennethgomez01, 2021-11-23:
@JernejL may I know if you have the genuine BLTouch? If so I am guessing maybe the probe is way too fast sending data back to the printer, try to disable fast probing in the firmware, and increase delay for sending data.

kennethgomez01, 2021-11-23:
@JernejL it is also possible the your bltouch is not able to get enough current to run its electronic parts, at which can cause the electric coil not able to push & retract the magnetic probe. Other reason would be the bltouch location is too close to heating element, this can cause damage to electric parts of the BLTouch.

kennethgomez01, 2021-11-23:
@JernejL it may also be a real problem with the firmware, cause in case the bltouch probe is too high touching the bed, it will trigger the "PROBE_Failed_Error", maybe try to connect to it to a computer and look at the serial to see what is happening when the error occur.

JernejL, 2021-11-24:
@kennethgomez01 true on many things: The bltouch is not original, but it ran fine for 5 months. it is positioned exactly where the instructions say to position it. 
 
After checking i conclude that the only problem could be with the pin header endstop wiring. I did it exactly as in instructions, but i think that dupont connector is unsuitable for use in the head assembly as it can vibrate around or cause bad connection - this makes sense with what i mentioned in https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/228 - if the whole printer is vibrating things are bound to go loose. 
 
I would suggest that installation instructions be changed, to recommend an interlocking JST connector or soldering wires directly to the pin header or routing them directly to the motherboard somehow in one piece. 
 
I'd stil like to understand why the probing procedure did not halt and crashed the head, this could be a pure marlin bug i guess? i could report it to them if this is so.

JernejL, 2021-12-08:
Still no idea? maybe it would be ok to report this to marlin directly?

kennethgomez01, 2021-12-08:
@JernejL that will be good idea, or I prefer to do it manually by installing the original marlin and do some changes according to your machine and using some code used in this Marlin raspository. I were you I should check the board, because it may be a wire problem maybe lost contact to the bltouch IC.

closed 2022-01-13 by github-actions[bot]