knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #230: [BUG] Filament in/filament out can not be stoped via TFT menu

vmax1145, 2021-11-23:
### Bug Description

Filament in and filament out with old blue-white TFT screen can not be stopped.

### Steps to Reproduce

1. Go to TFT menu  Tools->Filament on tft screen and press  Filament IN 
2. wait for extruder temperature reach 180 or more
3. Press Filament IN or Filament OUT 
4. Then Press Stop 

**Expected behavior:** Filament loading/unloading expected to be stopped immidiately. (as in stock firmware)

**Actual behavior:** Filament continue loading untill standard loading length step ends.

#### Additional Information
Printer: MEGA S, 
TFT screen: blue-white  , 
Motherboard: Trigorilla 1.1,  
firmware:  MEGA_S_v1.4.0-b3.hex  ( downloaded here )

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

vmax1145, 2021-12-24:
Any ideas how to fix this?

knutwurst, 2022-01-24:
Fixed in 1.4.0 Beta 5 ;)

closed 2022-01-24 by knutwurst