knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #234: [BUG] (short description)

foorb, 2021-12-04:
Hello,

First of all - thank you for your great work.

After a few months "break" I have started again with the 3d printing. I first updated cura (4.11), octoprint (1.7.2) and my Anycubic Mega S (1.3.1). Also i read the wiki again and done all steps.

When I start a print via cura using octoprint, I get the following error message:
`Failed to enable Bed Leveling`
This comes after the printer has heated up and moved to the start position.

I then tried the prusa-slicer. Here it comes to no error. However, I would like to switch back to cura.

Can you perhaps help me further?

### Bug Description

`Failed to enable Bed Leveling` after heat up and moving to start position

### Steps to Reproduce

Send print job via cura and octoprint

LeePorte, 2022-01-06:
I'm seeing the same thing this morning

LeePorte, 2022-01-07:
I had the following linex in my GCODE

```
 M420 S1                                    ; Enable leveling
 M420 Z5.0                                  ; Set leveling fading height to 5 mm
```
As I'd not created a levelling mesh yet (Testing driver change) it couldn't load the mesh. Removing these lines enabled printing to proceed

CruisenRainerWahnsinn, 2022-01-23:
On my 4MP2-DGUS (1.4.0b4) i can save mesh seetings to the eprom, but using "M420 S1 in start g code" causes an octoprint error: "Failed to enable Bed Leveling"

CruisenRainerWahnsinn, 2022-01-23:
```
M503
...
Recv: echo:; Mesh Bed Leveling:
Recv: echo:  M420 S0 Z0.00 ; Leveling OFF
Recv: echo:; Hotend PID:
...
[...]
Send: M420 S1
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N106601 M112*17
Send: N106602 M104 T0 S0*18
Send: N106603 M140 S0*87
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
```

knutwurst, 2022-01-24:
The bed leveling can only be activated when a mesh is stored. What's your complete output of M503?

CruisenRainerWahnsinn, 2022-01-24:
> The bed leveling can only be activated when a mesh is stored. What's your complete output of M503?

Ah, this might be causing the problem. I'll save some parameters, test again and report back. Thanks for the help and work you're putting in!

```
Recv: echo:; Linear Units:
Recv:   G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X100.00 Y80.00 Z800.00 E432.29
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X150.00 Y150.00 Z18.00 E30.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X1500.00 Y1500.00 Z70.00 E15000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P800.00 R1500.00 T1000.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.20 E10.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Mesh Bed Leveling:
Recv: echo:  M420 S0 Z0.00 ; Leveling OFF
Recv: echo:; Hotend PID:
Recv: echo:  M301 P30.96 I2.20 D109.06
Recv: echo:; Bed PID:
Recv: echo:echo:  M304 P251.78 I49.57 D319.73
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.20
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L538.00 U555.00 ; (mm)
```

knutwurst, 2022-01-24:
Great, that's good to hear!

So I'll close this now :) 

Best,
OIli

closed 2022-01-24 by knutwurst