knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #237: [BUG] Mega Pro (latest), inverted Z-Axis when using TMC firmware variant

jonas-koeritz, 2021-12-06:
### Bug Description

When I am using the TMC variant of the firmware (the device should come with TMC drivers and I have opened it and there seem to be TMC drivers in there) the Z-Axis moves inverted. If I home the printer, the head starts moving upwards insted of downwards.
I am currently running a print with the non-TMC enabled build and it's working perfectly (apart from whiney motors compared to stock firmware).

Did I miss anything? Any configuration option I can change at runtime?

### Steps to Reproduce

1. Upload `MEGA_P_DGUS_TMC_BLT_10_v1.4.0-b3.hex` to the printer.
2. Start homing

**Expected behavior:** Z-Axis moves towards the bed, into limit switches.

**Actual behavior:** Z-Axis moves upwards, away from the bed.

#### Additional Information

I am quite new to this, I really like that the laser option is coming to Marlin so I made the switch.

weihan-ott, 2021-12-07:
I got the same things.

I fixed it by these:

find:

  #if ENABLED(KNUTWURST_MEGA_P)
    // Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
    #define INVERT_X_DIR false // set to true for stock drivers or TMC2208 with reversed connectors
    #define INVERT_Y_DIR true // set to false for stock drivers or TMC2208 with reversed connectors
    #define INVERT_Z_DIR ture // set to false for stock drivers or TMC2208 with reversed connectors

change to

  #if ENABLED(KNUTWURST_MEGA_P)
    // Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
    #define INVERT_X_DIR false // set to true for stock drivers or TMC2208 with reversed connectors
    #define INVERT_Y_DIR true // set to false for stock drivers or TMC2208 with reversed connectors
    #define INVERT_Z_DIR ture // set to false for stock drivers or TMC2208 with reversed connectors

that's all.

jonas-koeritz, 2021-12-07:
Am I correct to say that [this line](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/0ae2f00fc8bddd03f74baa57ab9df33f55ac5181/Marlin/Configuration.h#L1846) isn't correct for a Mega_P with stock TMC2208 drivers, just if you upgraded them yourself because AnyCubic matches the wiring to account for the direction change when building the printer?

knutwurst, 2021-12-07:
@jonas-koeritz , no... it's for the TMC Build. Take a look at line https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/0ae2f00fc8bddd03f74baa57ab9df33f55ac5181/Marlin/Configuration.h#L1817 where the TMC is defined. :) 

If you own a stock Mega Pro, this is the line you are searching for: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/0ae2f00fc8bddd03f74baa57ab9df33f55ac5181/Marlin/Configuration.h#L1726

But... normally all users are fine with the configuration. It's strange, that your Z is inverted.

jonas-koeritz, 2021-12-07:
My stock mega pro comes with TMC Drivers, maybe that's the problem? It also
has a different PSU etc. so it seems to be another product revision?

knutwurst, 2021-12-07:
@jonas-koeritz the Stock pro comes with only 2 of 5 TMC drivers. The Non-TMC Firmware is always for the stock model. The TMC is always the correct FW if you change *all* drivers.

jonas-koeritz, 2021-12-07:
Ahh thanks for calcifying. So the marketing is just "we put some TMC
Drivers in it, so we can write it on the page", and I am doing everything
correctly (not running pure TMC Marlin).

knutwurst, 2021-12-07:
Correct ;) 
Just use the Non-TMC build if you own a stock mega Pro. If you decide to switch to TMC all the way, then you can take the _TMC Firmware.

jonas-koeritz, 2021-12-07:
Thanks for clarifying! Firmware work's like a charm, haven't tested the laser features yet ;).

closed 2021-12-07 by jonas-koeritz