knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #238: [BUG] Mesh leveling Z-axis off (1.4.0-b3)

JanJastrow, 2021-12-08:
### Bug Description

Hi, I think there is a bug in the current 1.4.0-b3 releases.
I was unable to correctly set my mesh z coordinates with this version (especially in the bottom-right & top-right corners)

### Steps to Reproduce

- Flashed MEGA_S_TMC_v1.4.0-b3 onto my Mega S
- Load FW defaults & save eeprom
- Restart
- Special Menu
- Prehead bed
- 4-point leveling
- mesh leveling

**Expected behavior:** [What you expect to happen]

Beeing able to reach the bed while mesh leveling

**Actual behavior:** [What actually happens]

Hovering above the bed

#### Additional Information

I was on version 1.2.0 before and just today upgraded to 1.4.0.-b3 to test the fancy new features.

I was always leveling my printer the way I always do: first 4-point and afterwards mesh.
4-point seems to be working as always, but when switching to mesh, I immediately noticed that I had to set way higher numbers then before - 0.1mm instead of just 0.01 or 0.02.
Still, I was able to set the first 4 points with the menu, but with the 5th point I was unable to get the nozzle down to the bed.
I tried multiple times & resets, but never succeded.

I've now downgraded to version 1.3.1 and everything is working as before on 1.2.0

knutwurst, 2022-01-24:
I'm on it... the leveling works, but there is an offset, some other users reported. I just need to figure out why :D

esprudzs, 2022-01-29:
Just my 5 cents. My config is very different from stock, so I did suspect it was due to some of the settings I have changed in the beginning (which could still be the case).

My symptoms are, that whatever the values are for mesh towards the 0,0 corner, positive or negative, depending on how I set the end-stop screws, the printing only lowers to about the probe offset (1.34 for me) above bed, Interestingly, this is gradually corrected towards the middle of the bed and the remaining three quadrants get reasonably well leveled (I can see Z axis doing the work).

knutwurst, 2022-02-15:
Should be fixed in >1.4.0

esprudzs, 2022-02-18:
>  after a toolchange (T1) and the thing is trying to print the layer at the same height or maybe even lower than there already is one, printed by T0.

I ran a print overnight on T1 only and got the same problem of "insufficient Z height". The nozzle was at 2.40 according to the screen, but at about 1.80 in reality. I'm printing with 0.16 layer height, so this is many layerheights off. My fade height is 0.96, but as I have seen before, the problem starts no later than at Z=0.64. I don't see anything wrong with slicing, so I will swap some hardware to rule that out.
[image]

knutwurst, 2022-02-24:
Is the issue still there?

esprudzs, 2022-03-29:
You can close this down. I worked around this issue, which is not to say leveling works - I think it still does not. But I have figured out how to work around it and I need to print, not to tinker for some time now :D. I hope to be able to upgrade to the latest firmware within a few weeks and lets see where we stand then. Cheers!

closed 2022-05-04 by github-actions[bot]