knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #245: [FR] Proper wiring

zerquetch, 2021-12-26:
after digesting the articles, there continues to be a problem with the wiring diagrams/oictures

If there was another way to post this, I would, but here it is:
I have a MegaPro. 
MB- TriG 0.0.2
Hotend PCB- Mega S V0.0.3 (X-Stop)

### Description
The informtion is inconsistant and I cannot find the correct solution to the wiring issues.
Browsing forums and 'the Web' lead to many more questions
Please inform of the correct wiring for an ABL using THIS hardware

So this is the pic listed in the issue but the information is seriously lacking
[image]

This shows the Black & White wires Splitting to 3 Pairs. Where are they going to?
The next pic says to swap pins
[image]
So, do we swap A5 and B5 since B6 also shows continuity on the B5 run? (coming from H1 &2) 
Which wires go into H1 & 2 exactly, White to Square Pin or Black?

I am happy to add schapps to your coffee fund

knutwurst, 2021-12-28:
Hi @zerquetch ;) 
Sorry for only answering now, but thanks to work and Christmas, there wasn't much time for the firmware and Github.

Regarding your question: It's a bit confusing, that's true, but basically it's very simple:
On every mega printer, pin 2 of the microcontroller is on pin B5 of the hotend circuit board. B6 (right next to it) always carries ground (GND).

Now there are different versions of the hotend circuit board, but the green plug on top is always the same.

If you follow the instructions you will find that a pin header is soldered to B5 and B6 of the Mega/S/X.

The Mega Pro already has a pin header for a sensor, so it is more practical to use it.

This connection leads to pins A5 and A6 / B6 (A6 and B6 are both ground / GND). But to put B5 and B6 on this connection at the end, you have to swap A5 and B5 in the green connector at the top. So then B5 and B6 are on the pin header of the Mega Pro.

By the way, the black wire of the BLTouch is ground / GND, which is why it must be connected to a GND pin. So A6 ... B6 ...

At the end (if you have swapped A5 and A6) all you have to do is plug in the plug of the BLTouch.

White on B5 (H1)
Black on B6 (H2)

This is your hotend pcb BEFORE the modification:
[image]

And this is how it should look like so you have both correct pins on H1 and H2:
[image]

I hope I can help you :)

Greetings and a happy new year,
Olli

Smack58078, 2022-01-01:
I have the same printer, but I have upgraded to a MK3 carriage. This causes the BLtouch to be in a different location. When probing, the BLtouch tries to probe off the bed. Can I get a firmware where the probe offset is changed. I can provide details. Just to verify, it is the Anycubic Mega Pro. 1.0 TMC

knutwurst, 2022-01-01:
You can change the probe offset via gcode: https://marlinfw.org/docs/gcode/M851.html

closed 2022-01-24 by knutwurst