knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #246: [BUG] starts beeping then lifts 20mm and stops

jeff222g, 2021-12-28:
### Bug description 

<!- - unable to continue from pause -->

### Steps to Reproduce

1. Press pause while printing, or have filament sensor be tripped
2. Attempt to resume print 
3. It sits there and does not do so

**Expected behavior:** [the print should resume from where it left off]

**Actual behavior:** [it will sit until you restart the printer]

#### Additional Information
This is happening on the mega pro without BL touch. I’m sorry this is my first time using GitHub

jeff222g, 2021-12-28:
It's happened with the filament sensor turned off as well. With the sensor turned off it just beeps with no error message and sits like you bit pause

jeff222g, 2021-12-28:
I now see the resume feature is not possible, but am not totally sure why my print is stopping itself. I'm trying to run a 30 hours print and am worried I will waste a lot of material on random stops

knutwurst, 2021-12-28:
Hi @jeff222g .. 
Are you sure you disabled the filament sensor before printing? it's only disabled temporary and will be re-enabled upon next boot.

The other question is: Which firmware version are you using? Start/Stop/Resume is not supported in the beta version atm.

jeff222g, 2021-12-28:
I'm not 100% sure whether or not I turned it back off after reboot, I know before starting the print a 3rd time I made sure to hit it, but I think I'm going to bypass the filament sensor entirely and just leave a cut piece of filament in it. I am using the beta version

makskr, 2022-01-05:
> 

Hi knutwurst . Which the last version supports pause/print for Chiron? 
I see that version 1.4.0-beta.3 has issues with filament sensor and pause/print.

knutwurst, 2022-01-24:
Fixed in 1.4.0 Beta 5.

closed 2022-01-24 by knutwurst